Lesserkuma/FlashGBX #1: [Feature Request/Help] Ability to add date and time to save backups

easthighNerd, 2020-11-14:
The default file names given to save backups do not contain the date and time of when the backup was created. Personally, I find it to be rather useful when periodically making backups of my games (i.e. Pokemon Crystal.)

I have been tinkering with the main Python script to attempt to add that functionality to it, and have succeeded in doing so all the time via
```
path = re.sub(r"[<>:\"/\\|\?\*]", "_", path) + "_" + datetime.now().strftime("%Y-%m-%d_%H-%M-%S") + ".sav"
```
, however I wanted to make it so that a variable can be read from the `config.ini` file (i.e. `AddDateTime=true`,) however I have thus far been unable to figure out how to get it to read that variable from the `config.ini` file.

I have looked over the file and have attempted using things such as
```
add_datetime = self.SETTINGS.value("AddDateTime")
```
to get it to read the variable, however I have not having any such luck.

---

Any insight into how I may be able to accomplish this while making minimal changes to the main Python file would be greatly appreciated!

I planned on making a pull request once I was able to get it ironed out so that hopefully others who might find this feature of interest would be able to benefit from it as well.

*- John*

lesserkuma, 2020-11-24:
Thanks for your suggestion, I will add it in the next version!
I think the problem is that QSettings ini files don’t seem to work well with boolean types, but I’ll look into it.

lesserkuma, 2020-12-17:
The feature was added in v0.9β.

closed 2020-12-17 by lesserkuma