Lesserkuma/FlashGBX #2: Windows Defender flags FlashGBX as a trojan horse

Siphonay, 2020-12-03:
Hey there, I’ve had FlashGBX suddenly close itself, and the executable getting automatically deleted and I couldn’t understand why. When I redownloaded it, it wouldn’t launch, I was getting a warning that it’s a malicious program. I just looked into Windows Defender’s logs and found this

[image]

I hope that’s a false negative and that you can fix it if you manage to reproduce it, because otherwise FlashGBX has been the best frontend for my GBxCart RW.

Have a nice day!

lesserkuma, 2020-12-05:
Hey, thanks for the report!
It seems like this is a widespread *false positive* issue with [PyInstaller](https://github.com/pyinstaller/pyinstaller/issues?q=is%3Aissue+virus+is%3Aclosed). I will try and compile the PyInstaller bootloader from source for the next release. I’m also planning to submit a report to Microsoft.

lesserkuma, 2020-12-17:
According to VirusTotal and Jotti’s Malware Scan, Windows Defender should no longer have a problem with the new version which was released today. I hope this fixes the issue for you as well!

closed 2020-12-18 by lesserkuma

Siphonay, 2021-01-12:
Sorry for the late reply but I can confirm I’m not having this issue anymore. Thanks for taking care of it!

Sheeg, 2026-09-08:
This issue appears to have returned for the windows x64 installer version of FlashGBX v5.1, Windows 11 flags it the zip file as a trojan and deletes it. The non-installer zip is fine.

[image]

Lesserkuma, 2026-09-08:
All that was changed in the build workflow was the upgrade to InnoSetup 7:
https://github.com/Lesserkuma/FlashGBX/commit/0272d8ef039b2e131a7278c1a3b6dd4ade5a7ecd#diff-a3ae20db8d6f829cce559bf09c6fa26bf26822b30ab4994444b49a0b9909470e

I guess I'll revert to the old version next time 🙁