Lesserkuma/FlashGBX #4: Backlog: Extremely slow under macOS

ClassicOldSong, 2020-12-17:


ClassicOldSong, 2020-12-17:
Interesting, using raspberry pi as a serial adapter it works much faster:

[image]

ClassicOldSong, 2020-12-17:
Commands on rpi:

```shell
modprobe dwc2
modprobe g_serial
stty -F /dev/ttyGS0 raw -echo
stty -F /dev/ttyAMA0 raw -echo 1000000
socat FILE:/dev/ttyGS0 FILE:/dev/ttyAMA0
```

lesserkuma, 2020-12-17:
I heard about different speeds on different machines, but I’m not exactly sure why that is.
Can you try to compile the [official GBxCart RW software](https://github.com/insidegadgets/GBxCart-RW) and measure its speed on the same machine?

ClassicOldSong, 2020-12-18:
The official one could not even open the ch340 device.

Interestingly enough, after I modified the program to recognize usbmodem, it works perfectly well with my rpi adapter.

lesserkuma, 2020-12-21:
> Interestingly enough, after I modified the program to recognize usbmodem, it works perfectly well with my rpi adapter.
Interesting, it does seem like performance partially depends on the driver. I also noticed serial drivers on different operating systems can have different input buffer sizes which may also be affecting performance and functionality; on Windows it seems to be at least 65536 bytes (didn’t test for more), on Ubuntu 4095 bytes and on macOS only 1020 bytes.

bbsan2k, 2021-01-04:
The CH34X kext that is provided with macos is pretty crappy... there is a (commercial) alternative driver, although the devs are currently struggling to establish support with Apples Driverkit. (https://www.mac-usb-serial.com/)

lesserkuma, 2021-01-22:
Thanks everyone! I’m closing this for now because I don’t think there is anything I could do if it's a driver-related issue.

lesserkuma, 2021-01-22:
Thanks everyone! I’m closing this for now because I don’t think there is anything I could do if it’s a driver-related issue.

closed 2021-01-22 by lesserkuma