Lesserkuma/FlashGBX #7: Can't install using Pip on Mac

teamsuperpanda, 2021-03-13:
On the Mac I get package versions conflicting using
`pip install FlashGBX`

```
`Collecting FlashGBX
  Using cached FlashGBX-1.4-py3-none-any.whl (263 kB)
Requirement already satisfied: setuptools in /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages (from FlashGBX) (49.2.1)
Collecting Pillow
  Using cached Pillow-8.1.2-cp39-cp39-macosx_11_0_arm64.whl (2.0 MB)
Collecting pyserial
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting FlashGBX
  Using cached FlashGBX-1.3-py3-none-any.whl (241 kB)
  Using cached FlashGBX-1.2.1-py3-none-any.whl (241 kB)
  Using cached FlashGBX-1.1-py3-none-any.whl (240 kB)
  Using cached FlashGBX-1.0-py3-none-any.whl (237 kB)
  Using cached FlashGBX-0.10-py3-none-any.whl (237 kB)
ERROR: Cannot install flashgbx==0.10, flashgbx==1.0, flashgbx==1.1, flashgbx==1.2.1, flashgbx==1.3 and flashgbx==1.4 because these package versions have conflicting dependencies.

The conflict is caused by:
    flashgbx 1.4 depends on PySide2
    flashgbx 1.3 depends on PySide2
    flashgbx 1.2.1 depends on PySide2
    flashgbx 1.1 depends on PySide2
    flashgbx 1.0 depends on PySide2
    flashgbx 0.10 depends on PySide2

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies`
```

lesserkuma, 2021-03-13:
Ah, the GUI portion of FlashGBX currently depends on `PySide2`. If the `PySide2` package is not available for your system, you can still install FlashGBX like so instead: `pip install pyserial pillow setuptools requests` followed by `pip install --no-deps FlashGBX`. This should allow you to use the command line interface of it. I'm sorry for the inconvenience and hope you will still find it useful.

I might try to add a second GUI using a different framework (Tkinter?) as a fallback some day if I have time.

teamsuperpanda, 2021-03-13:
Perfection. I misread the error and thought there were conflicting dependancies. I installed PySide2 as per your instructions and that solved that!

closed 2021-03-13 by teamsuperpanda