Lesserkuma/FlashGBX #10: "GUI mode couldn't be launched" with no error details

Skirmisher, 2021-04-22:
I've just installed FlashGBX on Fedora 33, by installing its dependencies via `dnf` and then installing the program via `pip install --user flashgbx`, which passed all dependency checks without downloading anything else:
```
Collecting flashgbx
  Downloading FlashGBX-1.4.2-py3-none-any.whl (264 kB)
     |████████████████████████████████| 264 kB 12.4 MB/s 
Requirement already satisfied: requests in /usr/lib/python3.9/site-packages (from flashgbx) (2.24.0)
Requirement already satisfied: pyserial in /usr/lib/python3.9/site-packages (from flashgbx) (3.4)
Requirement already satisfied: setuptools in /usr/lib/python3.9/site-packages (from flashgbx) (49.1.3)
Requirement already satisfied: PySide2 in /usr/lib64/python3.9/site-packages (from flashgbx) (5.15.2)
Requirement already satisfied: Pillow in /usr/lib64/python3.9/site-packages (from flashgbx) (7.2.0)
Requirement already satisfied: chardet<4,>=3.0.2 in /usr/lib/python3.9/site-packages (from requests->flashgbx) (3.0.4)
Requirement already satisfied: idna<3,>=2.5 in /usr/lib/python3.9/site-packages (from requests->flashgbx) (2.10)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /usr/lib/python3.9/site-packages (from requests->flashgbx) (1.25.8)
Installing collected packages: flashgbx
Successfully installed flashgbx-1.4.2
```

However, the program always launches in CLI mode, with the message regarding "GUI mode couldn't be launched" but with no further details. I'm certainly not missing core Qt functionality, seeing as I'm using the Plasma DE. How can I debug this further?

lesserkuma, 2021-04-22:
Hi, thanks for the issue report!
I asked around and this command may resolve the issue on Fedora: `dnf install python3-pillow-qt`
Can you give it a try?

Skirmisher, 2021-04-22:
Thanks, that did the trick! And now I realize that the description for the `python3-pillow` package describes that there are four subpackages including `pillow-qt`, which I skimmed over...

closed 2021-04-22 by Skirmisher