Lesserkuma/FlashGBX #12: Could not find a version that satisfies the requirement

bad-dad-games, 2021-05-28:
Following the instructions (installing pip through the terminal on Linux Mint) then using pip install FlashGBX I got the following message...

>  Could not find a version that satisfies the requirement FlashGBX (from versions: )
No matching distribution found for FlashGBX

What do I do next? I assume it's something to do with pointing the console in the right direction. Many thanks.

lesserkuma, 2021-05-28:
Hi! Maybe you can try `pip -V`. It should say `(python 3.x)` at the end of the line. If it doesn’t, try to reinstall Python 3 and pip for Python 3.

bad-dad-games, 2021-05-28:
Many thanks
> Hi! Maybe you can try `pip -V`. It should say `(python 3.x)` at the end of the line. If it doesn’t, try to reinstall Python 3 and pip for Python 3.
That's the problem I got this...
> pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)

lesserkuma, 2021-05-28:
Oh ok, FlashGBX was developed in Python 3 and it won’t work with Python 2.
After installing Python 3 and its pip, it’s possible you have to use `pip3 install FlashGBX` instead of `pip install FlashGBX`.

bad-dad-games, 2021-05-28:
Should I uninstall the older versions of pip and Python? I'm still having the same issue. I have in my usr/lib folders from Python2.7 - 3.8.

lesserkuma, 2021-05-28:
Shouldn’t be necessary to uninstall older versions. What is the error message now?
Are you using these commands?
`pip3 install FlashGBX`
`python3 -m FlashGBX`

bad-dad-games, 2021-05-28:
I'm still getting the 
_could not find a version..._
I re installed pip, it says I have the latest but is still using the (python 2.7) version.

bad-dad-games, 2021-05-28:
I just tried `pip3 -V` and it's using (python 3.6). So I then tried `pip3 install FlashGBX` but I still got `could not find a version that satisfies...`

lesserkuma, 2021-05-28:
I *think* Python 3.6 is too old as the package is labeled to require Python 3.7+.

bad-dad-games, 2021-05-28:
What is limiting my pip3 to python3.6? Do I need to do a system update?

LucentW, 2021-05-28:
> What is limiting my pip3 to python3.6? Do I need to do a system update?
If you're still on Tricia (Linux Mint 19.3), you need to update at least to Ulyana (Linux Mint 20) which is based on [Ubuntu Focal Fossa](https://packages.ubuntu.com/focal/python3) to have Python 3.8.2.

bad-dad-games, 2021-05-28:
Many thanks, I really appreciate your help. I will close this issue.

closed 2021-05-28 by bad-dad-games

bad-dad-games, 2021-05-30:
Solution!
For people having PIP problems, where `pip -V and pip3 -V` only indicate versions of Python before 3.7. Install python 3.7 onto your system then use `/usr/bin/python3.7 -m pip install FlashGBX` to install and `/usr/bin/python3.7 -m FlashGBX` to run. This got round the problem of potentially breaking my system by changing the default pip.