Lesserkuma/FlashGBX #17: YJencrypted Issues

RibShark, 2021-08-20:
These carts should be fully dumpable with the right techniques:

- First the cart needs to be "initialised" with the exact/similar reads to a GBA, parsing the header ideally on the first read (some bytes in the Nintendo logo are replaced if the header is read a second time, and sometimes will contain trap zones too).
- Second, the cart needs the addresses to be read in roughly the same way as the game does. I am unsure of the details here, it could be as simple as first reading the initial jmp, following to that location and reading that, or it could be more complex requiring a certain level of emulation to follow the right path. Doing this "unlocks" reading the rest of the cart, which would otherwise stop reading entirely at around ~0x100 (IIRC).
- In most of these carts, there are "trap zones" which, upon reading, will prevent any further reads of the cart. From my experience, these come in two types: a 4-byte long zone, of which there are many, usually near the start of the address space, and a large chunk (typically 0x10000 bytes or more) of which reading any bytes will prevent further reads (returning either 0x00 or a repeating pattern of 4 bytes). Not all YJencrypted cards have these areas (my Mario Cart cart doesn't, for instance), but most do. Once a trap zone has been hit, the cart will need to be powered off and reinitialised to read any further.
- The full 32mb address space needs to be dumped, as this cart appears to mirror the ROM in unpredictable ways that are unique to each cart, and the code *will* jump to addresses in the mirrored areas.

Right now, FlashGBX fails at the start, refusing to even read the header of any of my carts (Sonic 3 Fighter Sonic, Crash Advance IV, Rayman IV and Mario Cart). Since your firmware doesn't appear to be open source, I can't say for certain what is going wrong here, but it is possible the AGB_BOOTUP_SEQUENCE command does not simulate the cart reads correctly. There could also be some factor I am not aware of outside of reads affecting this, such as timing.
For the trap zones, the best way to do it would be to power on/off the device whenever a trap zone is detected (can be done by attempting to read an earlier address and comparing values), though I do not believe every USB hub allows powering off like that.

lesserkuma, 2021-08-20:
Hey there, thanks for the helpful message!

I was always curious about the YJencrypted carts, do you know where to get one cheap, maybe even from within the EU?

The AGB_BOOTUP_SEQUENCE command only exists since firmware version L2 for the GBxCart RW v1.4 hardware revision. Which hardware and firmware versions are you using?

Also, with the GBxCart RW v1.4 hardware we can now power cycle a cartridge without the need of disconnecting the USB connection.

RibShark, 2021-08-20:
Ah, I do not have a v1.4, so I wasn't aware of the ability to power cycle, that sounds extremely helpful in this case. I'm using a v1.3 with the L1 firmware. Now that I know that the command doesn't exist on that firmware, I may try and see if I can get at least the header to read with just AGB_CART_READ commands.

I obtained all my carts via saved searches on ebay for the various Sintax GBA bootlegs (all of which are YJencrypted carts), I think they all came from the UK or Germany.

lesserkuma, 2021-08-20:
I see. The GBxCart RW v1.3’s MCU does not have enough flash memory for further firmware improvements, that’s why there is currently no L2 for the v1.3. Using AGB_CART_READ from within FlashGBX should work, it’d just be slower.

I couldn’t find any Sintax carts on eBay today, but I will keep checking.

RibShark, 2021-08-20:
Seems that the cart reads I was using that worked on the GBA hardware do not work with the GBxCart. Not sure if the hardware itself does anything special aside from those reads, or whether the cart requires faster timing to work. I will try and order a v1.4 to test with once it becomes in-stock at Zedlabz.

ClassicOldSong, 2021-10-08:
It seems that I could get some random YJEncrpyted carts on the market but I'm not sure if there just clones of clones. I could try to read these carts with my own dumb designed reader to see if it can read just fine.

RibShark, 2021-10-08:
> 
> 
> It seems that I could get some random YJEncrpyted carts on the market but I'm not sure if there just clones of clones. I could try to read these carts with my own dumb designed reader to see if it can read just fine.

I have never heard of clones of carts (such as Sintax bootlegs) that are known to be YJencrypted, the point of the encryption was to prevent clones like this and it seems they managed to.

That being said, we have been unable to replicate the boot sequence necessary to unlock the cart even with the help of lesserkuma's logic analyser, but if you think you could crack it then that would be brilliant.

closed 2026-10-31 by Lesserkuma