Lesserkuma/FlashGBX #19: Trouble reading cartridge header on Linux

shinyquagsire23, 2021-09-10:
Recently got a GBxCart RW v1.4 Pro and tried out the latest version, but I cannot for the life of me get it to read the cartridge header.

I messed around a bit on my own and found that if I send 0xC1 commands in a loop, it will eventually read chunks of 512 bytes, but otherwise it will read nothing and I can't figure out if there's a way to get the device to just flush the data. Waiting longer doesn't do anything and as soon as I ^C, the data shows up in Wireshark. Can't tell if it's a Linux driver being weird and buffering inputs or what.

Console Output:
```
[maxamillion@ShinyQuagsire-To-Go FlashGBX]$ sudo python3 run.py --debug --mode agb --action backup-rom
FlashGBX v2.8 by Lesserkuma
https://github.com/lesserkuma/FlashGBX
[2021-09-09 20:28:37] [/home/maxamillion/workspace/FlashGBX/FlashGBX/Util.py:97] SetValue(): Updating settings: ConfigVersion = v2.8
Now running in CLI mode.

Configuration directory: /home/maxamillion/workspace/FlashGBX/FlashGBX/config

[2021-09-09 20:28:37] [/home/maxamillion/workspace/FlashGBX/FlashGBX/hw_GBxCartRW.py:146] Initialize(): Firmware information: {'cfw_id': 'L', 'fw_ver': 2, 'pcb_ver': 5, 'fw_ts': 1628497800, 'fw_dt': '2021-08-09T01:30:00-07:00', 'ofw_ver': 31}
[2021-09-09 20:28:37] [/home/maxamillion/workspace/FlashGBX/FlashGBX/hw_GBxCartRW.py:147] Initialize(): Baud rate: 1700000

[2021-09-09 20:28:37] [/home/maxamillion/workspace/FlashGBX/FlashGBX/hw_GBxCartRW.py:146] Initialize(): Firmware information: {'cfw_id': 'L', 'fw_ver': 2, 'pcb_ver': 5, 'fw_ts': 1628497800, 'fw_dt': '2021-08-09T01:30:00-07:00', 'ofw_ver': 31}
[2021-09-09 20:28:37] [/home/maxamillion/workspace/FlashGBX/FlashGBX/hw_GBxCartRW.py:147] Initialize(): Baud rate: 1700000

For help please visit the insideGadgets Discord: https://gbxcart.com/discord

Connected to GBxCart RW v1.4 – Firmware R31+L2 (/dev/ttyUSB0) (dated 2021-08-09T01:30:00-07:00)
Cartridge Mode: Game Boy Advance
[2021-09-09 20:28:37] [/home/maxamillion/workspace/FlashGBX/FlashGBX/hw_GBxCartRW.py:391] _set_fw_variable(): Setting firmware variable ADDRESS to 0x0
[2021-09-09 20:28:37] [/home/maxamillion/workspace/FlashGBX/FlashGBX/hw_GBxCartRW.py:605] ReadROM(): Reading 0x180 bytes from ROM at 0x0 in 6 iteration(s)
[2021-09-09 20:28:37] [/home/maxamillion/workspace/FlashGBX/FlashGBX/hw_GBxCartRW.py:391] _set_fw_variable(): Setting firmware variable TRANSFER_SIZE to 0x40
[2021-09-09 20:28:37] [/home/maxamillion/workspace/FlashGBX/FlashGBX/hw_GBxCartRW.py:391] _set_fw_variable(): Setting firmware variable ADDRESS to 0x0
[2021-09-09 20:28:38] [/home/maxamillion/workspace/FlashGBX/FlashGBX/hw_GBxCartRW.py:378] _read(): Error: Received 0 byte(s) instead of the expected 64 byte(s)
bytearray(b'')
```

lsusb -v info:
```
Bus 001 Device 022: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1a86 QinHeng Electronics
  idProduct          0x7523 CH340 serial converter
  bcdDevice            2.64
  iManufacturer           0 
  iProduct                2 USB Serial
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      1 
      bInterfaceProtocol      2 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

lesserkuma, 2021-09-10:
Hi there!
It seems like the recent erroneous driver patch could be the problem. It was only recently reverted, but your distribution may still use the broken driver version. Also see https://bugzilla.kernel.org/show_bug.cgi?id=214131

shinyquagsire23, 2021-09-15:
OK yep, updated my kernel and it's been fixed, thanks!

closed 2021-09-15 by shinyquagsire23