Lesserkuma/FlashGBX #20: Compatibility with 2006_TSOP_64BALL_QFP48

Sithdown, 2021-09-23:
I've received a reproduction Game Boy cartridge with this version: 2006_TSOP_64BALL_QFP48

[image]

I'm not sure FlashGBX is able to flash to this cartridge. Any tips?

Sithdown, 2021-09-23:
Hi!

It's a bootleg GBC Pokemon Yellow (Spanish Version) from Aliexpress. It's from this seller:
[aliexpress.com/item/1005001537927786.html?spm=a2g0s.9042311.0.0.656f63c0hdcHDQ](aliexpress.com/item/1005001537927786.html?spm=a2g0s.9042311.0.0.656f63c0hdcHDQ)

I was curious because I haven't seen this PCB on previous cartridges. It seems every seller has its own version.

I'm waiting for my FlashGBX to arrive (1-2 weeks), so I'll be able to get more info then. I hope we can learn something about this version by then!

Thank you for your time :)

Sithdown, 2021-10-04:
Hello again!

I've finally checked the cartridge with FlashGBX and GBxCart RW 1.4, firmware version R31+L2.

[image]

When doing a ROM backup, the checksum is not correct.

If I Auto-detect the flash cartridge, it looks like it's compatible:

[image]

So it seems it's a "SD007_BV5_V3 with AM29LV160MB".

Is it safe to write a ROM on this flashcart? I don't want to brick it that soon 😅

lesserkuma, 2021-10-04:
Oh cool, glad it works! It seems to work exactly the same as the four others that are currently listed by FlashGBX, so I will add yours to that list soon, probably going to call it “2006_TSOP_64BALL_QFP48 with AL016J55FFAR2”. Thanks again!

Sithdown, 2021-11-20:
Resolved by d2abd30

closed 2021-11-20 by Sithdown