Lesserkuma/FlashGBX #21: As for the problem of diyGBC burning card chip

dyf2007, 2021-12-08:
As for the problem of diyGBC burning card chip,
[image]
[image]
[image]
[image]
[image]
# I use AM29F016B-75SI and use general to burn normal games. I hope that god can add this chip. The following picture is the chip data. Thank you for updating this software.

dyf2007, 2021-12-08:
[image]

lesserkuma, 2021-12-09:
Very nice! Can you write the ROM successfully with your newly added setting?

dyf2007, 2021-12-09:
> Very nice! Can you write the ROM successfully with your newly added setting?

Can, the game can be written correctly and effect, the game can play normally, half a year ago this burning card is done, has been modified with this burning game, tried a lot of games。

lesserkuma, 2021-12-09:
Thank you for your contribution, I will add it to the next release based on your pictures!

closed 2021-12-10 by lesserkuma