Lesserkuma/FlashGBX #22: Compatibility with Flash2Advance 128M Flash Card

twitnic, 2021-12-23:
Is it possible to add this card to FlashGBX?

[image]

[image]

```
The following cartridge configuration was detected:

Game Title: GB_Lo_er

Flash ID Check:
[   ROM   ] 2E 00 89 00 24 FF 17 00 
[   555/AA] 89 00 00 EA 17 00 AE 51 
[   555/A9] 89 00 00 EA 17 00 AE 51 
[  5555/AA] 89 00 00 EA 17 00 AE 51 
[  5555/A9] 89 00 00 EA 17 00 AE 51 
[  4555/AA] 89 00 00 EA 17 00 AE 51 
[  4555/A9] 89 00 00 EA 17 00 AE 51 
[  7555/AA] 89 00 00 EA 17 00 AE 51 
[  7555/A9] 89 00 00 EA 17 00 AE 51 
[     0/90] 89 00 89 00 17 00 17 00 

Common Flash Interface Data: No data provided

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (0/90)” a try.

FlashGBX v3.2 | GBxCart RW v1.4 – Firmware R33+L4 (/dev/cu.usbserial-1430)
```

lesserkuma, 2021-12-23:
Hi! Yeah, I think I can add this cartridge. As there are different revisions of the Flash2Advance cartridges, would it be possible for you to open the cartridge and take a picture of the circuit board with readable chip markings?

twitnic, 2021-12-25:
[image]

[image]

I hope that helps.

lesserkuma, 2021-12-26:
Yes that helps and it seems to be very similar to the 256M cartridge that I own, so this might work!

1. Click [Config] → Show configuration directory
2. Download and copy [fc_AGB_Flash2Advance_128M.txt](https://github.com/lesserkuma/FlashGBX/files/7776955/fc_AGB_Flash2Advance_128M.txt) into the folder that opened
3. Restart FlashGBX

With this, the software should be able to auto-detect your cartridge. Then please test if you can successfully write a 16 MB ROM file!

twitnic, 2021-12-31:
It works. Sorry I was busy (Christmas Holidays .. )

I put the txt file into the config folder and tried to write a rom to the cartridge. it was successfully :)

Thanks !

[image]
[image]

lesserkuma, 2022-01-03:
I’m glad to hear that! It will be added to the next release version as well :) Thank you!

closed 2022-01-03 by lesserkuma