Lesserkuma/FlashGBX #23: New Cart or what else?

x7l7j8cc, 2022-01-11:


LucentW, 2022-01-11:
Do you have a GBxCart to poll the CFI off that cartridge or at least the flash ID?
The flash chip looks like it has been sanded, so with no markings @lesserkuma can't really find the datasheet for that chip and implement it in FlashGBX :(

x7l7j8cc, 2022-01-11:
@LucentW I've inserted the photo from the phone first, then i continued to write from the computer.

LucentW, 2022-01-11:
Well, even though the cartridge types don't match the PCB markings, 99.9% you should be able to flash it with any of those profiles.
Quite strange that the flash chip is 2 MB only on a batteryless save cart -- usually the bootleggers opt for 4 MB because of the chips' large sector size (and obviously the same sector erase size).

You can contribute PCB pics on the [FlashcartDB](https://flashcartdb.com/index.php/Main_Page) Wiki and @lesserkuma will update the supported carts list with the markings on your cart when you confirm that it flashes cleanly.

x7l7j8cc, 2022-01-11:
@LucentW I've already write that i flashed successfully a game, the game flashed was Pokemon Crystal Clear.

Plays very smoothly, and i think this might have the SRAM onboard but no battery pin as far i can see.
Because when i save and soft-reset there still the save there, but if i turn off completely my GameBoy the save is gone.

LucentW, 2022-01-11:
That cart is too small to fit Crystal Clear and the batteryless patch (for the reasons explained before -- there's no empty 0x10000 bytes segment + a bank for the read/save routines to sacrifice), you need a 4 MB cart for that.
If you want to go for the battery backed solution, Perfect Dark and Pokémon TCG2 carts are best bet since both are 4 MB ROMs and the bootleggers don't bother patching them, so you should also get a cart with a moderately power efficient SRAM chip, battery retainer and preinstalled CR2032.

x7l7j8cc, 2022-01-11:
@LucentW  Obviously i mean a Clean Pokemon Crystal Clear (not patched).
And this because i opened after i flashed the ROM to see inside. 

I dont want to patch the game, i want to add a battery XD If i can. There's plenty room to put a battery.
But first i need to find the correct 3V pin for the SRAM.

lesserkuma, 2022-01-12:
That cartridge looks like it was not designed to work with a battery as there are no battery terminals.

closed 2022-01-17 by lesserkuma