Lesserkuma/FlashGBX #25: Board with unlisted model number

mrsonicblue, 2022-01-19:
I have a board with a model number which isn't listed. When trying similar model numbers, the ROM won't write successfully. I can't tell if it's incompatible or is broken.

[image]

lesserkuma, 2022-01-19:
Hi! Can you provide more information?
- What game was originally stored on this cartridge?
- Can you post a higher quality picture of the small chip on the left so the text on it can be read?
- Can you post what FlashGBX says when you click [Detect Flash Cart]?

mrsonicblue, 2022-01-19:
Sure. I'll be able to do that tonight (about 10 hours from now).

mrsonicblue, 2022-01-20:
@lesserkuma Here is the information you asked for. The original game was Mario Kart Super Circuit. But it was crashing at the title screen, so it never actually worked.

Clearer photos of the two chips:

[image]
[image]

And the flash cart details:

```
Game Title: MARIO KART
ROM Size: 16 MB
Save Type: 512K SRAM (64 KB)

Flash ID Check:
[   ROM   ] 06 00 10 EA 24 FF AE 51 
[   AAA/A9] 20 00 7D 22 00 00 08 00 

Common Flash Interface Data:
Swapped pins: (0, 1)
Device size: 0x0800000 (8.00 MB)
Voltage: 2.7–3.6 V
Single write: True
Buffered write: True (32 Bytes)
Sector erase: 1024–8192 ms
Sector flags: As shown (0x02)
Region 1: 0x0000000–0x000FFFF @ 0x2000 Bytes × 8
Region 2: 0x0010000–0x07FFFFF @ 0x10000 Bytes × 127

Compatible Cartridge Types:
- 2006_TSOP_64BALL_6106 with W29GL128SH9B*
- AGB-E05-01 with GL128S
- AGB-E05-01 with MSP55LV128M
- AGB-E05-02 with JS28F128
- AGB-E05-02 with M29W128FH
- AGB-E05-02 with M29W128GH
- BX2006_0106_NEW with S29GL128N10TFI01
- BX2006_TSOP_64BALL with GL128S
- BX2006_TSOPBGA_0106 with M29W640GB6AZA6

FlashGBX v3.3 | GBxCart RW v1.4 – Firmware R33+L4 (COM3)
```

mrsonicblue, 2022-01-20:
If I use this model: BX2006_TSOPBGA_0106 with M29W640GB6AZA6

When I read the ROM, The data is just `EE 00 AE` repeating forever.

[image]

mrsonicblue, 2022-01-20:
Another observation. I found another game using the same model of board. It's Advance Wars 2, which also crashes on the title screen on a clone system (Analogue Pocket). But it runs fine on a Gameboy DS.

lesserkuma, 2022-01-20:
I see! It looks like the maximum ROM size for this cartridge would be 8 MB, not 16 MB. Can you try to write a smaller ROM and see if that works?

mrsonicblue, 2022-01-20:
@lesserkuma I tried setting it to 8MB without success. It fails almost immediately, at the beginning, when writing data that's different than what's on the cart. If I write the same data that's already on the cart, it "succeeds". Presumably because the verification then works. But only when set to 8MB, as you suggested. If set to 16MB, it fails right after 8MB (as you'd expect).

lesserkuma, 2022-01-20:
Sorry, I’m not sure why it’s not working with your cartridge. You could try the [Config] → Prefer full chip erase option and see if that makes a difference.

mrsonicblue, 2022-01-20:
@lesserkuma I'll try the full chip erase that you suggested tonight. If you have any interest, I can send you one of these carts. You won't need to send it back. It's not important to make this work, so it's up to you. If not, I will close this issue.

closed 2022-01-24 by mrsonicblue