Lesserkuma/FlashGBX #26: Compatibility with another SD007_TSOP_48BALL_V10 cartridge

howie0210, 2022-02-01:
[image]

Would be great is support for this cart could be added. None of the current profiles work 

SEC 
KM62256CLTC-5

AMD
L160DB12V 0108FPG U

LucentW, 2022-02-01:
Can you please do a `Detect Cartridge` and post the details here? It'll greatly help @lesserkuma into adding support for your flash chip.

howie0210, 2022-02-01:
ROM Size: 2 MB
Save Type: 256K SRAM (32 KB)

Flash ID Check:
[     ROM     ] F3 E0 FD 3E 00 C3 C9 00 
[WR   / AAA/A9] 02 02 4A 4A 00 00 00 00 
[WR   /4AAA/A9] 02 02 4A 4A 00 00 00 00 
[WR   /7AAA/A9] 02 02 4A 4A 00 00 00 00 

Common Flash Interface Data:
Swapped pins: (0, 1)
Device size: 0x0200000 (2.00 MB)
Voltage: 2.7–3.6 V
Single write: True
Buffered write: False
Sector erase: 1024–16384 ms
Region 1: 0x0000000–0x0003FFF @ 0x4000 Bytes × 1
Region 2: 0x0004000–0x0007FFF @ 0x2000 Bytes × 2
Region 3: 0x0008000–0x000FFFF @ 0x8000 Bytes × 1
Region 4: 0x0010000–0x01FFFFF @ 0x10000 Bytes × 31

Compatible Cartridge Types:
- 2006_TSOP_64BALL_QFP48 with AL016J55FFAR2*
- SD007_BV5_V3 with AM29LV160MB
- SD007_TSOP_29LV017D with L017D70VC
- SD007_TSOP_48BALL with AM29LV160DB
- SD007_TSOP_48BALL with AM29LV160DT
- SD007_TSOP_48BALL with L160DB12VI
- SD007_TSOP_64BALL_SOP28 with unlabeled flash chip

FlashGBX v3.3 | GBxCart RW v1.4 – Firmware R33+L4 (COM9)

lesserkuma, 2022-02-02:
Hi, what is the exact error message? This might work: Choose the `Generic Flash Cartridge (WR/AAA/A9)` option from the Cart list and then click [Write ROM] and [Use 5V].

howie0210, 2022-02-02:
WR/AAA/A9 at 5V works, it doesn’t at 3.3v

The auto detect and most other profiles will timeout “ Erasing the flash
chip timed out.
Please make sure that the
cartridge contacts are clean, and
that the selected cartridge type
and settings are correct.”

Only message in the console is “ NOTE: This cartridge's Flash ID (F3 E0 FD
3E) doesn't match the cartridge type selection.”

lesserkuma, 2022-02-03:
Okay! I will add the voltage selection selection for this cartridge type in the next release version.

closed 2022-02-04 by lesserkuma