Lesserkuma/FlashGBX #27: mac osx

imun0ne, 2022-02-17:
hey my friend, first of all. thx for the great great work.
im not really into programming, more into art and photo. 
so i don't understand at all, where and how to download the macosx version of flashgbx.
maybe you can help me out.
thx&greets

lesserkuma, 2022-02-17:
Hello! Did you follow the [directions](https://github.com/lesserkuma/FlashGBX#run-using-python-linux-macos-windows)? Basically download the Python installer from https://www.python.org/downloads/macos/ and run it, then open up a Terminal window (From the Finder menu bar: Go → Utilities → Terminal). Next enter `pip3 install -U FlashGBX` to download and install the macOS version of FlashGBX. Finally enter `python3 -m FlashGBX` to run the application. You can also join the [insideGadgets Discord](https://www.gbxcart.com/discord) where other macOS users will be able to help you out.

imun0ne, 2022-02-17:
Thx for the quick response. Perfect description. Download&Install worked fine. Now i just need to figure out how to download photos from my gameboy camera...

imun0ne, 2022-02-17:
Thx my friend you helped me a lot. everything working fine now! Only question left, do i have run the install everytime i wanna use flashgbx on my mac?

closed 2022-02-18 by lesserkuma