Lesserkuma/FlashGBX #28: Wrong insideGadgets cart detected

johnny-pumpernickel, 2022-02-24:
Hello,

After updating to FlashGBX v3.5, my 4 MB insideGadets cart can only be used as a 2 MB cart.

I have an insideGadgets 4 MB (S29GL032M) flash cart. In previous versions, I could manually select it and flash without a problem. After upgrading, when I run "Detect Flash Cart" it is detected as an insideGadgets 2 MB, 128 KB SRAM/32 KB FRAM cart. If I manually select my proper cart and try to write my game file, I get the following errors:

```
CFI ERROR: Couldn’t retrieve buffer size from the cartridge.
CFI ERROR: Couldn’t retrieve buffer size from the cartridge.
NOTE: This cartridge’s Flash ID () doesn’t match the cartridge type selection.
CFI ERROR: Couldn’t retrieve sector size map from the cartridge.
CFI ERROR: Couldn’t retrieve sector size map from the cartridge.
Waiting for confirmation from the device has failed. (Called from WriteROM(), line 1091)

Flash error at 0x4000 in iteration 0 of 8 while trying to write a total of 0x2000 bytes (response = False)
```
Let me know if you need any more information.

Thank you

johnny-pumpernickel, 2022-02-27:
Now that I think about it, I'm not sure it ever detected the cart correctly except for the first time. Once I started flashing my game to it, it would detect a different cart. But, I could manually choose "insideGadgets 4 MB (S29GL032M)" and it would work.

A photo of the PCB is here:
(https://imgur.com/PQoIEZx)

The Detect Flash Cart details:
```
Game Title: RETRO TRIVIA DU
ROM Size: 2 MB
Save Type: 256K SRAM (32 KB)

Flash ID Check:
[     ROM     ] FF FF FF FF FF FF FF FF 
[WR   / AAA/AA] 01 01 D8 D8 00 00 00 00 

Common Flash Interface Data:
Device size: 0x0200000 (2.00 MB)
Voltage: 4.5–5.5 V
Single write: True
Buffered write: False
Sector erase: 1024–8192 ms
Region 1: 0x0000000–0x0003FFF @ 0x4000 Bytes × 1
Region 2: 0x0004000–0x0007FFF @ 0x2000 Bytes × 2
Region 3: 0x0008000–0x000FFFF @ 0x8000 Bytes × 1
Region 4: 0x0010000–0x01FFFFF @ 0x10000 Bytes × 31

Compatible Cartridge Types:
- insideGadgets 2 MB, 128 KB SRAM/32 KB FRAM*

FlashGBX v3.5 | GBxCart RW v1.4 – Firmware R33+L4 (/dev/ttyUSB0)
```

lesserkuma, 2022-03-10:
Thanks for the photo and info. It should work if you update to v3.6 and manually select this option:
`insideGadgets 4 MB (2× 2 MB), 32 KB FRAM, MBC5`

johnny-pumpernickel, 2022-03-13:
Thank you for the patch! I updated to v3.6 and was able to successfully flash 4mb games again by manually selecting the new option you added.

Your program is really cool! I love being able to test my homeboy on a real cartridge.

closed 2022-03-13 by johnny-pumpernickel