Lesserkuma/FlashGBX #29: Add Delete ROM Feature

daidianren, 2022-02-25:
request to add this feature

lesserkuma, 2022-03-05:
OK, I’m planning to add this feature in the next release.

lesserkuma, 2022-03-09:
In v3.6 you can click [Write ROM] and then cancel the file select dialog to trigger the new feature.

closed 2022-03-09 by lesserkuma

daidianren, 2022-03-15:
> 在 v3.6 中，您可以单击 [Write ROM]，然后取消文件选择对话框以触发新功能。
Glad to add a hidden feature to this, so proud of you! cheer!
When I extract the rom, it prompts me that the map hidden sector file is missing, what is this file? Is it the game map file? Where should this file be placed? I am using a Mac system and a Nintendo white 8M flash card. For this flash card, I know there is a special program. I have also used GBxCart_RW_Nintendo_Power_Cart_Programs_v1.6, but there are too few options, so I can't really burn it perfectly. Recording LSDJ, because the 128K SRAM cannot be used perfectly, the songs cannot be saved. I have successfully burned LSDJ on the modified EZ FLASH 2 burner before, and it can be saved perfectly. Now the price of an EZ FLASH 2 burner Expensive and very hard to find, drivers are also a big problem, so again I request that in the near future, I can do some optimizations for this burn card so that it can burn LSDJ perfectly, thank you again for your reply, thank you for making contribute!
My English is not very good, most of the content here is google translated, I hope you can understand my question.

lesserkuma, 2022-03-15:
I generated a hidden sector (.map) file for LSDj v9.2.L, can you give it a try?
[GB Memory (LSDj-v9.2.L) [8F+16B].zip](https://github.com/lesserkuma/FlashGBX/files/8252879/GB.Memory.LSDj-v9.2.L.8F%2B16B.zip)

daidianren, 2022-03-15:
> 我为 LSDj v9.2.L 生成了一个隐藏扇区 (.map) 文件，你可以试试吗？ [GB内存（LSDj-v9.2.L）[8F+16B].zip](https://github.com/lesserkuma/FlashGBX/files/8252879/GB.Memory.LSDj-v9.2.L.8F%2B16B.zip)
succeeded
I have been trying for a week, every day until 7 am, and I have to go to work at 12 noon, which almost affects my normal life.
How is it done? Can I update LSDJ normally afterwards?
Really appreciate your help!

lesserkuma, 2022-03-15:
Awesome, I’m glad it works! I wasn’t aware that none of the usual tools can generate this file for 128 KB of SRAM, so I manually created the file for you. Maybe a future version of FlashGBX will be able to generate this file automatically.
And yes, you will be able to use the same .map file with future versions of LSDj.

daidianren, 2022-03-15:
> 
Thank you for your great contribution, looking forward to adding this feature to future versions of FlashGBX, thanks again for your help! thank you very much!