Lesserkuma/FlashGBX #31: Flash Advance Extreme => Same Flash Chips as Flash 2 Advance 128M

schmingy, 2022-03-08:
This is maybe less of an "issue" and more of a question. I have a Flash Advance Extreme cartridge that has the same flash chips as the Flash 2 Advance 128M listed as compatible on the site. The FA Extreme cartridge still works with a GBA and the Flash2Advance cartridge I have reads with FlashGBX just fine.

Is there something special about the "Extreme" cartridge that FlashGBX repeatedly disconnects when attempting to read the cartridge?

lesserkuma, 2022-03-08:
Hi! I do not know this cartridge. Does it disconnect when just _reading_ the cartridge? What buttons are you clicking exactly? If it disconnects even when just reading the header data, that could mean that there’s a short circuit somewhere and the polyfuse of the GBxCart RW device tripped. Are there any error messages, maybe also in the terminal output?

schmingy, 2022-03-08:
I plug in the reader, then launch the software, then click Game Boy Advance. The software appears to try to read and then throws the below error message:

[image]

Terminal output - after clicking Yes once and then No.
FlashGBX v3.5 by Lesserkuma
https://github.com/lesserkuma/FlashGBX

You are using the latest version of FlashGBX.

Connected to GBxCart RW v1.4 – Firmware R33+L4 (COM3) (dated 2021-11-21T07:15:30-06:00) at 1.7M baud
For help please visit the insideGadgets Discord: https://gbxcart.com/discord
Disconnected from GBxCart RW v1.4 – Firmware R33+L4 (COM3)

Connected to GBxCart RW v1.4 – Firmware R33+L4 (COM3) (dated 2021-11-21T07:15:30-06:00) at 1.7M baud
For help please visit the insideGadgets Discord: https://gbxcart.com/discord
Disconnected from GBxCart RW v1.4 – Firmware R33+L4 (COM3)
Traceback (most recent call last):
  File "package\FlashGBX\FlashGBX_GUI.py", line 1375, in SetMode
  File "package\FlashGBX\FlashGBX_GUI.py", line 1399, in ReadCartridge
AttributeError: 'NoneType' object has no attribute 'ReadInfo'

Below is a photo of the cartridge board itself.
[image]

schmingy, 2022-03-08:
Using the insideGadgets official software with the Flash Advance Extreme cartridge, the software attempts to read it and hung to where I had to kill it through the task manager. With the other F2A card, it worked just fine.

The cartridge does play fine in a GBA. What I haven't done yet is try the cartridge with the original Visoly Flash Advance linker and software to see if I can reflash it through that. It took quite a bit of fiddling to get that software to work on a VM running WIndows 98 since the software is so old, but I did manage to salvage the save data I had from the F2A card. Perhaps I'll give that a go and see if the cart just is dead to alteration and just can work with what it has on it now. I've had these since I bought them originally from LikSang back in the early 2000s.

lesserkuma, 2022-03-08:
It does sound like the device is protecting itself from a short circuit, so any software loses the connection. I’m not sure what could be causing this in your case, but you can try to align the cartridge’s pins very carefully with the GBxCart RW’s cart slot. Also make sure there’s no conductive dust between the pins (maybe clean the pins with IPA on a cotton swab). If it still doesn’t work, maybe Alex from insideGadgets knows what else you could try. You can reach him on the iG Discord as @/AlexiG#8193.

lesserkuma, 2022-03-09:
When the disconnect happens, do you see any LEDs go off? Also what happens if you push the physical button on your GBxCart RW to power on the cartridge (Power LED should go on) and then run the software?

lesserkuma, 2022-03-09:
I just released a new version of FlashGBX which also includes a new firmware update (version R34+L5). Can you try it again with that?

schmingy, 2022-03-10:
Thanks, I just tried it. The behavior was a little different this time, although instead of just connecting, disconnecting, and giving up, it kept retrying.

Here is a portion of the log as I didn't think it necessary to copy the repeating disconnections after the first few.

FlashGBX v3.6 by Lesserkuma
https://github.com/lesserkuma/FlashGBX

This version of FlashGBX (3.6) seems to be newer than the latest release (3.5). Please check for updates manually.

Connected to GBxCart RW v1.4 – Firmware R33+L4 (COM3) (dated 2021-11-21T07:15:30-06:00) at 1.7M baud
For help please visit the insideGadgets Discord: https://gbxcart.com/discord
Disconnected from GBxCart RW v1.4 – Firmware R33+L4 (COM3)

Connected to GBxCart RW v1.4 – Firmware R34+L5 (COM3) (dated 2022-03-09T12:12:25-06:00) at 1.7M baud
For help please visit the insideGadgets Discord: https://gbxcart.com/discord
Disconnected from GBxCart RW v1.4 – Firmware R34+L5 (COM3)

Connected to GBxCart RW v1.4 – Firmware R34+L5 (COM3) (dated 2022-03-09T12:12:25-06:00) at 1.7M baud
For help please visit the insideGadgets Discord: https://gbxcart.com/discord
Disconnected from GBxCart RW v1.4 – Firmware R34+L5 (COM3)

Connected to GBxCart RW v1.4 – Firmware R34+L5 (COM3) (dated 2022-03-09T12:12:25-06:00) at 1.7M baud
For help please visit the insideGadgets Discord: https://gbxcart.com/discord
Disconnected from GBxCart RW v1.4 – Firmware R34+L5 (COM3)

Connected to GBxCart RW v1.4 – Firmware R34+L5 (COM3) (dated 2022-03-09T12:12:25-06:00) at 1.7M baud
For help please visit the insideGadgets Discord: https://gbxcart.com/discord
Disconnected from GBxCart RW v1.4 – Firmware R34+L5 (COM3)

lesserkuma, 2022-03-10:
Does the disconnect also happen when you hotplug the cartridge? Start the software without any cartridge, click Game Boy Advance and Read Info, then insert the cartridge and click Read Info again.

schmingy, 2022-03-10:
Same basis result. I did notice that the same light sequence I described earlier happened when the unit was set up as you described as soon as the cartridge was plugged into the reader immediately but stopped after one cycle. When I clicked Read Info, the same disconnect and reconnect kept taking place.

lesserkuma, 2022-03-17:
Unfortunately I think there’s nothing I can do in my software to make it work 😞... Maybe if you could lend Alex from iG one of these, he might be able to figure this out on a hardware level.

closed 2023-01-10 by lesserkuma