Lesserkuma/FlashGBX #35: Can't detected AGB-E02-20

aszzq, 2022-04-16:
Pokémon LeafGreen Version(JP)
[image]
[image]
[image]

lesserkuma, 2022-04-17:
Hi, can you try again with the latest version of FlashGBX?
You can find it here: https://github.com/lesserkuma/FlashGBX/releases/latest

If it still doesn’t work, try to clean the cartridge contacts very thoroughly. You can also try not to push the cartridge all the way into the cart reader. I tested my Japanese Pocket Monsters Leaf Green cartridge and it works ok on FlashGBX v3.7 using GBxCart RW v1.3 on firmware L1.

[image]
[image]

aszzq, 2022-04-18:
> Hi, can you try again with the latest version of FlashGBX? You can find it here: https://github.com/lesserkuma/FlashGBX/releases/latest
>
> If it still doesn’t work, try to clean the cartridge contacts [...]

Hi, I have cleaned the cartridge contacts contacts with alcohol，then used FlashGBX v3.7.
IT still doesn’t work.

[image]

lesserkuma, 2022-04-18:
Thanks for checking. In that case, I don’t think the problem is related to my software. It seems to be a connectivity problem or a hardware issue with either the cartridge or the GBxCart RW device.

You could change the firmware to version R30 and try again with insideGadgets’ official software: https://shop.insidegadgets.com/wp-content/uploads/2018/05/GBxCart_RW_GUI_Interface_v1.49.zip
If it still doesn’t work, maybe contact insideGadgets customer support.

aszzq, 2022-04-20:
> Thanks for checking. In that case, I don’t think the problem is related to my software. It seems to be a connectivity problem or a hardware issue with either the cartridge or the GBxCart RW device. [...]

Not working.
I send it to someone who have the game for testing.

aszzq, 2022-07-18:
Something wrong with the PCB what I drew. I redraw the PCB，That's OK.

closed 2022-07-19 by aszzq