Lesserkuma/FlashGBX #39: Can't backup save data from SD007_TS0P_64BALL_S0P28

fabiodelgadopereira, 2022-05-28:
I can't backup save data from **SD007_TS0P_64BALL_S0P28**

Which option do I select in the combobox "**save type**" and "**mapper type**"?

[image]
[image]

LucentW, 2022-05-28:
Hi Fábio,
that cartridge is using batteryless saving, so it means that the SRAM chip there is actually empty and used only as a staging area so that the routines the bootleggers added to the game reflash the SRAM into the game ROM.

You need to dump the entirety of the ROM (do a Detect Flash Cart first to recognize the real flash size), pop it into an emulator with a debugger like BGB to identify the ROM location where the save is, then extract it manually using an hex editor. You can then edit it and paste it back into the same location again and reflash it to the cart to change the save.

fabiodelgadopereira, 2022-05-29:
Thanks! Using BGB I was able to export the save

closed 2022-05-30 by lesserkuma