Lesserkuma/FlashGBX #43: Any version past 1.4.2 refuses to write faster than 0.11 KB/s

twadepsvita, 2022-07-09:
I'm running this on macOS and I've been stuck using version 1.4.2 for a while. Any time I try a new version, it refuses to flash roms faster than 0.11 KB/s. 1.4.2 doesn't have this weird issue. Thought I should report it to you. Thanks.

lesserkuma, 2022-07-13:
Hi, what is the cart type under which you are trying to write ROMs? Also please provide the GBxCart RW hardware revision and firmware version numbers.

twadepsvita, 2022-07-13:
> Hi, what is the cart type under which you are trying to write ROMs? Also please provide the GBxCart RW hardware revision and firmware version numbers.

GBxCart RW v1.3 Pro with latest firmware and the below cart from InsideGadgets, which was the newest version at the time of purchase (February 2021).
GBA 32MB, 1Mbit Flash Save with RTC, Flash Cart (Works with Pokemon games)

lesserkuma, 2022-07-13:
Oh there are two different firmware versions for the v1.3 Pro, which one are you using? Version R30 or L1? Please try if the other of the two works better. FlashGBX can update the firmware through Tools → Firmware Updater.

twadepsvita, 2022-07-14:
> Oh there are two different firmware versions for the v1.3 Pro, which one are you using? Version R30 or L1? Please try if the other of the two works better. FlashGBX can update the firmware through Tools → Firmware Updater.

I'll try this now and let you know how it goes.

closed 2022-07-15 by lesserkuma