Lesserkuma/FlashGBX #46: FreeBSD support

Efiartes, 2022-07-18:
I ma trying to use cli version of FlashGBX but in my FreeBSD13.1 system it always says:
```
No devices found.
```

The device is detected as follows in dmesg:
```
ugen0.10: <vendor 0x1a86 USB Serial> at usbus0
uchcom0 on uhub1
uchcom0: <vendor 0x1a86 USB Serial, rev 1.10/81.33, addr 12> on usbus0
uchcom0: CH340 detected
```

And a new device is created:
```
ls -la /dev/cua*
crw-rw----  1 uucp  dialer  0x13e Jul 18 10:55 /dev/cuaU0
crw-rw----  1 uucp  dialer  0x13f Jul 18 10:55 /dev/cuaU0.init
crw-rw----  1 uucp  dialer  0x140 Jul 18 10:55 /dev/cuaU0.lock
```

If you need some more testing contact with me, i will be pleased to help you.

Efiartes, 2022-07-18:
I have uninstalled pip version:
```
pip uninstall pyserial
```

And installed with packages:
```
pkg install py39-pyserial-3.5_2
```

But same result:
```
Select Operation:
 1) Read Cartridge Information
 2) Backup ROM
 3) Write ROM
 4) Backup Save Data
 5) Restore Save Data
 6) Erase Save Data
 7) Extract Game Boy Camera Pictures From Existing Save Data Backup
 8) Firmware Update (for GBxCart RW v1.4/v1.4a only)

Enter number 1-8 [1]: 1
No devices found.
```

lesserkuma, 2022-07-18:
Doesn’t seem to be an issue with FlashGBX directly, but you can try a few different things!
- Use a different USB cable (many cheaper ones do not have the data lines connected)
- Use a different USB port on your computer
- Avoid using USB hubs
- Try a different computer running on a different operating system and see if it works there
If it still doesn’t work, maybe contact insideGadgets customer service.

LucentW, 2022-07-18:
The serial identification is done by pySerial and not FlashGBX, that's why I'm pointing towards it having some quirks on FreeBSD.

Relevant lines of code: https://github.com/lesserkuma/FlashGBX/blob/14d401a3f95c6fc738919698a0bd00f68bd8612c/FlashGBX/hw_GBxCartRW.py#L126-L130

PySerial documentation confirms that it doesn't fully populate the fields on *BSDs, but only on Windows/Linux/macOS: https://pyserial.readthedocs.io/en/latest/tools.html#serial.tools.list_ports.comports

@lesserkuma you might have a dropdown and let the user type in the device node or a CLI parameter to point out the path to skip the auto stuff? :thinking:

lesserkuma, 2022-07-18:
Alright, too bad pySerial doesn’t fully work on *BSDs. Will add a command line argument to override the device path in the next version.

LucentW, 2022-07-18:
Probably it's trying to use `/dev/cuau0` instead of `/dev/cuaU0`? `termios` erroring out is usually because it can't set parameters on the port.

Efiartes, 2022-07-18:
Executing it as root, i get same error.
Can i test /dev/cuau0 manually using netcat or some similar software?
What should i expect coneccting to /dev/cuau0, some kind of string?

Efiartes, 2022-07-19:
I have tried but without luck:
```
			comports = serial.tools.list_ports.comports()
			for i in range(0, len(comports)):
				print(comports[i])
				#if comports[i].vid == 0x1A86 and comports[i].pid == 0x7523:
				#	ports.append(comports[i].device)
				ports.append(comports[i].device)
			ports = [ "/dev/cuaU0" ]
			if len(ports) == 0: return False
```

No devices found.

lesserkuma, 2022-07-31:
I added command line argument `--device-port` to override the device path to v3.16.1. If the problem still remains, I think it can likely only be fixed by pyserial maintainers.

closed 2022-07-31 by lesserkuma