Lesserkuma/FlashGBX #47: Compatibility with Megamemory

gdureuil, 2022-07-25:
Hello,

I recently bought a GBxWriter 1.4a and use it for some of my games with FlashGBX.

I have a suggestion regarding the Megamemory cart. It's like an Action Replay, pluging between the gameboy and your cartridge, and enabling you to backup and restore saves on actual carts. It has no battery (can't run dry) and was of great use before GBxCart.

Going by this wiki page - https://tcrf.net/Mega_Memory_Card_(Game_Boy) - it uses SST 28SF040 EEPROM/flash memory. I can't find it in the list and can't seem to dump the data bank.

Is it possible to dump the big databank of the Megamemory ? Would it be possible to isolate individual saves in that databank and extract them as proper saves ?

Thanks for your time.

gdureuil, 2022-07-26:
If it can help you, here are internal photos of Megamemory.

[image]
[image]

lesserkuma, 2022-07-26:
It looks a bit like a GameShark from the inside. If I ever come across one myself, I will take a look.

happysoul, 2023-10-23:
27C010 = 128KB
28SF040 = 512KB
rom total = 640KB

20V8H is a CPLD like chip, after program it use for 8 input 8 output (or maybe 10in 10out . i forgot it... )
work with HC373 chip to switch chip and chip pin address

you can only rewrite chip with a programmer ( chip 27c010 need 12.5V for earse/write , 28sf040 need 5v read/write )