Lesserkuma/FlashGBX #48: Squareboi Flashcart Compatibility

ALXCO-Hardware, 2022-07-29:
Hello! 

I am the developer for Squareboi, an open-source GB/GBC flashcart. I would like to ask if it could be added to the list of supported cartridges for future versions of FlashGBX.

The flashcart behaves transparently as either an MBC1 or MBC5 cartridge, depending on jumper settings, and has a maximum capacity of 2MB ROM/32KB RAM, and 4MB ROM/32KB RAM in those modes, respectively. The ROM write line is tied to cartridge pin 31 (VIN/AUDIO).

It can be assembled with a variety of ROM ICs, both in DIP and TSOP-48 for factors. Here is the list of currently supported ones:

DIP: SST39SF010A, SST39SF020A, SST39SF040

TSOP-48: M29F200FB, M29F400FB, M29F800FB, M29F160FB

RAM ICs, meanwhile, are limited to FM18W08 and FM1808B.

There is also an option to have two M29F160FB ICs working in tandem, with a demultiplexer to choose between them, in order to expand the total capacity to 4MB.

Here are a few other details:

-Due to the usage of FRAM, cartridge pin 2 (Clock) must be held Low when reading or writing to RAM. Additionally, the CS line must be strobed High in between consecutive RAM operations, to allow for a pre-charge period.

-When using the full 4MB memory option, the second flash IC is only active when ROM Address line 21 is High. This means that the addresses of commands sent to that chip must be offset by 0x200000, so that they are located at the start of memory bank 128.

For further documentation, and the complete source code of the current Arduino-based cart reader implementation I am currently using, please check the Squareboi github project page, [Here.](https://github.com/ALXCO-Hardware/squareboi)

I am also happy to provide any details or clarification that might be missing from the docs, as they are still a work in progress.

Thank you in advance for your consideration!

lesserkuma, 2022-07-31:
Hi! Since I do not have any of your cartridges at hand, it’s a bit difficult for me to add support for all these myself, but you could add support yourself by creating .txt files for them. You can find the existing ones at [Tools] → Open configuration directory. Try to experiment with the values and then submit those that work for your cartridges and I can add them to the next release! Use the [Detect Flash Cart] button to get some of the required information such as Flash ID values.

ALXCO-Hardware, 2022-08-01:
Okay! I will study the config files and try to make ones for my flahscart.

At the moment, I don't have access to a GBxCart, so I'll try to acquire one to verify my work before submitting. It might take a little while.

lesserkuma, 2023-01-10:
Closing for now, please re-open if necessary.

closed 2023-01-10 by lesserkuma

ALXCO-Hardware, 2023-01-10:
Will do! I'll assemble the config files when manual bank switching on AUDIO carts is implemented.