Lesserkuma/FlashGBX #53: EZ Flash V/3-in-1 Support Request

TankedThomas, 2022-11-10:
Hopefully this is the right place for this.
I was wondering if you could please add support for the EZ Flash 3-in-1 (which is labelled EZ Flash V, but it's a GBA/Slot 2 card, not the NDS one). 
It has 256Mb NOR Flash for storing a single ROM so you can use it in a GBA instead of having to use a DS, but you've typically needed a DS to flash it. Since I don't have a DS any more, it would be nice if I could write to it with the GBxCart.

I tried it with a GBxCart v1.4a and the latest FlashGBX (3.19), but it can't even detect the cartridge at all - it just errors out after a while of searching. I've only tried it on Windows so far but I can try it on my Mac or on Linux too if necessary. 
I have an ancient (pre-Joey) BennVenn dumper which can dump the ROM and save from this cart (and they work in an emulator), but it doesn't have write support. 
Since I don't have a DS any more, it would be nice if I could write to it with the GBxCart.

I'm happy to help debug issues with it if you'll add support for it. I just don't want to have to buy another DS only for this cart.

This is the error log from cmd, for what it's worth (seems pretty useless to me though):
```
Traceback (most recent call last):                                                                                        
  File "package\FlashGBX\FlashGBX_GUI.py", line 1473, in SetMode                                                          
  File "package\FlashGBX\FlashGBX_GUI.py", line 1486, in ReadCartridge                                                    
  File "package\FlashGBX\FlashGBX_GUI.py", line 1421, in CheckDeviceAlive 
```
This repeats several times, and then ends with:
```
  File "package\FlashGBX\FlashGBX_GUI.py", line 1417, in CheckDeviceAlive
  File "package\FlashGBX\FlashGBX_GUI.py", line 701, in FindDevices
  File "package\FlashGBX\hw_GBxCartRW.py", line 135, in Initialize
  File "serial\serialwin32.py", line 33, in __init__
  File "serial\serialutil.py", line 244, in __init__
  File "serial\serialwin32.py", line 80, in open
  File "serial\serialwin32.py", line 126, in _reconfigure_port
ctypes.ArgumentError: argument 2: <class 'RecursionError'>: maximum recursion depth exceeded while calling a Python object
```

Here's a photo of the cartridge too, just in case:
[image]

lesserkuma, 2022-11-10:
Are you on the currently latest firmware version which is R37+L7? If yes, can you try if it works better with the [official software directly from insideGadgets](https://shop.insidegadgets.com/wp-content/uploads/2018/05/GBxCart_RW_GUI_Interface_v1.49.zip)?

TankedThomas, 2022-11-10:
Sorry, I forgot to add the firmware version. Yes, I'm using the latest version R37+L7.

I did try the official software (mentioned) but it does nothing whatsoever.
The most I can get out of it is this error when clicking "Read ROM":
[image]

Also, I tried FlashGBX on my Mac (running Monterey 12.6.1) but it did the same as Windows.

Oh, and since the cart has an open front, I did make sure the pins are all aligned.

insidegadgets, 2022-11-11:
Hi there,

Perhaps the cart is drawing too much power at once that it's tripping the hardware. 

Could you please connect a wire between these 2 points?
[image]

And if that doesn't work, you can try to bypass the green fuse with the label "bg" but keep a close eye on the board since a component may start to get hot if there is a short circuit somewhere when the cart is inserted.

TankedThomas, 2022-11-12:
Sure thing.

Before soldering, I tested at those two points whilst trying to read the cartridge, and the voltage only ever reached 3.15V. It was 3.18V at idle. Not sure if this is relevant, but thought I would mention it.

After putting a wire between them both, it does indeed work! However, it reads it as a generic cartridge which isn't rewritable - that's a problem. Screenshot below is the message I got when using "Detect Flash Cart".
[image]

It seems to dump the save and ROM correctly, but it refuses to write to it because it detects it as unwritable.

My first question is if I should leave this jumper wire in place or not.
My next one, obviously, is if/how there is a way around this.

Also, this might not be anything useful, but I took the EZ Flash 3-in-1 out of its shell and found that it has some pins on the rear. No idea what they're for, but clearly they don't work in a GBA.
[image]

insidegadgets, 2022-11-14:
It could just have been that the cart draws too much current at start up and thus is tripping the part we bypassed. I've only experienced this if I purposely made a cart that had some pins shorted out together. You could leave it as is but you lose the cart power on/off functionality we have via the software or button.

Likely the pins on the back are programming pins for the CPLD or FPGA on board. That device could also lock out any re-flashing of the cart unless special commands are issued to it.

TankedThomas, 2022-11-14:
So do you think it's fixable (preferably without the wire in-place)? Perhaps something I should ask Alex about? This isn't really in my wheelhouse but I'd like to get it working if possible, and I'm happy to put in some work to achieve that.

I find it interesting that it is designed to work in a Nintendo DS, as that would also be running 3.3V to the cart, and there are a handful of tools available that work with it (such as GBAExploader or Rudolph's 3in1 Expansion Pack Tool). Surely then, there would be a way to get it to work with the GBxCart and other dumpers? Unless the DS is capable of providing more than 3.3V to the GBA slot, perhaps.

So it's definitely rewritable, and without the original software (which I assume was just the EZ Flash V software for the Slot 1/DS card).
But I tried with the official GBxCart Flasher tool, and I couldn't get it to work at all. Also, erasing/writing a save only works by changing the default option (after reading) in FlashGBX to `Unlicensed 512K SRAM (64KB)`.

I tried all the options, and the most I could get was that most of the 32MB options would allow me to begin the writing process, but it would get stuck trying to erase the cartridge and thus not go any further (and the ROM + save on the cart is still intact afterward, though one of the options managed to corrupt the save at some point).
The main errors are: 
* `Waiting for sector erase has timed out. Please unplug GBxCart RW, re-seat the cartridge and try again.` 
(This shows the writing progress bar but then times out with this error).
* `Waiting for chip erase has timed out. Please unplug GBxCart RW, re-seat the cartridge and try again.` 
(This shows "Erasing..." but eventually times out).
* `Flash chip doesn't appear to be responding. Please re-seat the cart and power cycle GBxCart.` 
(This happens before the writing process begins).
* `Writing has timed out. Please unplug GBxCart RW, re-seat the cartridge and try again.`
(This only occurred with the `16MB (Nintendo AGB Cartridge 128M Flash S, E201850)` option).

Oh, and the auto-detect option thinks that it's 4MB NOR Flash (it already has a 32MB ROM on it, so it's definitely not, and I haven't used it much, so it shouldn't be deteriorated to the point that it's unwritable).

Don't know if it's any help but I did get these IDs:
* When trying to flash to it via the flasher tool:
```
Read ROM: 0x2E, 0x0, 0x0, 0xEA, 0x24, 0xFF, 0xAE, 0x51,
Flash ID (AAA, A9): 0x2E, 0x0, 0x0, 0xEA, 0x24, 0xFF, 0xAE, 0x51,
Flash ID (AAA, AA): 0x2E, 0x0, 0x0, 0xEA, 0x24, 0xFF, 0xAE, 0x51,
Flash ID (00, 90): 0x2E, 0x0, 0x0, 0xEA, 0x24, 0xFF, 0xAE, 0x51,
```
* Via GBxCart RW Flash ID Check (GB):
```
Read ROM: 0x2, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID (555, AA): 0x2, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID (555, A9): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID (AAA, AA): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID (AAA, A9): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID (AA, 98):  0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID (7AAA, A9): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID Pin (5555, AA): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID Pin (5555, A9): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID Audio Pin (555, AA): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID Audio Pin (555, A9): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID Audio Pin (AAA, AA): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID Audio Pin (AAA, A9): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID Pin (5555, AA): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Flash ID Pin (5555, A9): 0xF0, 0x0, 0x4D, 0x33, 0x0, 0x0, 0x0, 0x0, 
Finished  
```
* Via GBxCart RW Flash ID Check (GBA):
```
Read ROM: 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
Flash ID (AAA, A9): 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
Flash ID (AAA, AA): 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
Flash ID (555, A9): 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
Flash ID (555, AA): 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
Flash ID  (00, 90): 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
Finished
```

lesserkuma, 2022-11-14:
The source code you found may indeed be useful to add support to FlashGBX! However, before trying to port this over, the cartridge must be usable without bypassing safety components on the GBxCart RW and I don’t know how that can be achieved… Will have to wait and see if @insidegadgets comes up with a solution.

TankedThomas, 2022-11-14:
Sounds good. I don't know if they're easy to get any more - apparently they're somewhat expensive - but just in case, there are multiple versions of the same cart as noted here: https://gbatemp.net/threads/how-to-tell-which-version-of-the-3in1-you-have.130659/

I don't really want to send mine to anyone in case I never get it back (even with the best intentions, things go awry sometimes), but if it's absolutely necessary, I could probably lend it to either of you, if that's really the only viable option.