Lesserkuma/FlashGBX #55: Retry on failed sector write

corwinkuiper, 2022-11-15:
When attempting to write a rom to a cartridge, it fails after some random time. I've seen it write 2MB without error before failing, but at some point with large roms it fails.
[image]

If rather than restarting the whole process, if it could retry to write the failed writes then I think it should increase the likelihood of writing the whole rom.

lesserkuma, 2022-11-30:
I released the new version today. I hope it works for you!

closed 2022-11-30 by lesserkuma

corwinkuiper, 2022-12-01:
Works flawlessly, thanks very much!
[image]