Lesserkuma/FlashGBX #56: Support for Flash2Advance 128M with 4xM6MGT323S4TP

magiblot, 2022-12-03:
Hi! I have this cartridge which FlashGBX doesn't recognize. It does not have a game stored. Do you think support can be added?

Thank you very much!

```
The following cartridge configuration was detected:

Game Title: 

Flash ID Check:
[   ROM   ] FF FF 1C 00 FF FF 20 00 
[   555/AA] 1C 00 1C 00 20 00 20 00 
[   555/A9] 1C 00 1C 00 20 00 20 00 
[  5555/AA] 1C 00 1C 00 20 00 20 00 
[  5555/A9] 1C 00 1C 00 20 00 20 00 
[   AAA/AA] 1C 00 1C 00 20 00 20 00 
[   AAA/A9] 1C 00 1C 00 20 00 20 00 
[  AAAA/AA] 1C 00 1C 00 20 00 20 00 
[  AAAA/A9] 1C 00 1C 00 20 00 20 00 
[  4555/AA] 1C 00 1C 00 20 00 20 00 
[  4555/A9] 1C 00 1C 00 20 00 20 00 
[  7555/AA] 1C 00 1C 00 20 00 20 00 
[  7555/A9] 1C 00 1C 00 20 00 20 00 
[  4AAA/AA] 1C 00 1C 00 20 00 20 00 
[  4AAA/A9] 1C 00 1C 00 20 00 20 00 
[  7AAA/AA] 1C 00 1C 00 20 00 20 00 
[  7AAA/A9] 1C 00 1C 00 20 00 20 00 
[     0/90] 1C 00 1C 00 20 00 20 00 

Common Flash Interface Data: No data provided

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (0/90)” a try at your own risk.

FlashGBX v3.20 | GBxCart RW v1.4a – Firmware R38+L8
```

[image]
[image]

lesserkuma, 2022-12-03:
Hi! I have yet to get one of these quad/octo F2A carts in my hands before I can add support for them. There’s no harm if you try the `Flash2Advance 128M (with 2× 28F640J3A120)` option, but it probably won’t work... 😢

magiblot, 2022-12-03:
No, that option does not work. Is there anything I can do to attempt this myself, or to let you access my machine remotely?

lesserkuma, 2022-12-03:
Ah remote debugging doesn’t work for me unfortunately. Would need to hook this up to a logic analyzer (e.g. DSLogic U3Pro32) and capture unlock sequences and write commands while using the original flasher. If you can provide such a capture file, that might help!

magiblot, 2023-01-07:
It looks like the original linker is a device that connects to the GBA and uses the GBA itself to flash the cart. I may be able to find the program that is sent to the GBA; would that help?