Lesserkuma/FlashGBX #61: DIY MBC3 flashcard FRAM Save Data support

KoopsF, 2022-12-21:
I've modified my DIY MBC3 flashcard by replacing the SRAM with FRAM and the mod seems to work. I tested it with Pokemon Prism and the game can save to the chip and it holds the save data even when I remove the battery.

But I noticed FlashGBX does not want to read/write properly to the new FRAM. Is it possible to get support for writing and backing up save data from FRAM kits like this?

RTC2FRAM kit found on [Aliexpress](https://www.aliexpress.com/item/1005003330733047.html)
FRAM chip [FM18W08](https://www.infineon.com/dgdl/Infineon-FM18W08_256-Kbit_(32_K_8)_Wide_Voltage_Bytewide_F-RAM_Memory-DataSheet-v06_00-EN.pdf?fileId=8ac78c8c7d0d8da4017d0ec89b424117)

[image]

[image]

lesserkuma, 2023-01-10:
I have very little experience with hardware stuff, but all FRAM carts I have work. It may be faulty FRAM or something else entirely. Maybe someone on insideGadgets’ discord may be able to help: https://gbxcart.com/discord

closed 2023-01-10 by lesserkuma

KoopsF, 2023-01-12:
Thank you for your comment. 
I think you might be right. I am going to try another FRAM chip.

lesserkuma, 2023-01-12:
I often hear about broken FRAM chips. I hope you can get it working! If not, please drop by on Discord :)