Lesserkuma/FlashGBX #62: Whats the best way to update FlashGBX on linux?

tolik518, 2022-12-30:
I ran the `pip3 install "FlashGBX[qt5]"` which generated the following output:
```
Requirement already satisfied: FlashGBX[qt5] in ./.local/lib/python3.8/site-packages (3.14)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from FlashGBX[qt5]) (45.2.0)
Requirement already satisfied: pyserial in ./.local/lib/python3.8/site-packages (from FlashGBX[qt5]) (3.5)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from FlashGBX[qt5]) (2.22.0)
Requirement already satisfied: python-dateutil in /usr/lib/python3/dist-packages (from FlashGBX[qt5]) (2.7.3)
Requirement already satisfied: Pillow in /usr/lib/python3/dist-packages (from FlashGBX[qt5]) (7.0.0)
Collecting PySide2; extra == "qt5"
  Downloading PySide2-5.15.2.1-5.15.2-cp35.cp36.cp37.cp38.cp39.cp310-abi3-manylinux1_x86_64.whl (164.3 MB)
     |████████████████████████████████| 164.3 MB 96 kB/s 
Collecting shiboken2==5.15.2.1
  Downloading shiboken2-5.15.2.1-5.15.2-cp35.cp36.cp37.cp38.cp39.cp310-abi3-manylinux1_x86_64.whl (975 kB)
     |████████████████████████████████| 975 kB 1.5 MB/s 
Installing collected packages: shiboken2, PySide2
Successfully installed PySide2-5.15.2.1 shiboken2-5.15.2.1
```

Which looks like it updated _something_ but not FlashGBX itself. I had FlashGBX 3.14 installed, and it seems like it hasn't updated to 3.20. So whats the preferred way to update FlashGBX?

LucentW, 2022-12-30:
Just add `--upgrade` before `FlashGBX[qt5]`.

tolik518, 2022-12-30:
> Just add `--upgrade` before `FlashGBX[qt5]`.

Thanks a lot, that worked like a charm. It would be nice to add this to the readme I think.

closed 2022-12-30 by tolik518