Lesserkuma/FlashGBX #64: Cartridge not holding Pokemon Crystal Clear save

Bullet55NL, 2023-01-04:
Hi! All I have a cartridge with a TC58FVT016F Flash chip, with a SD007_T40_64BALL_TSOP28 PCB text. I just wrote a copy of Pokemon Crystal Clear on this and everything works perfectly until I save and shutdown Gameboy Color. After I completely shutdown my GBC the save file is gone and I have to start a new game.. Any ideas? Am I doing something wrong writing the save type?

lesserkuma, 2023-01-04:
Hi! If it’s a bootleg cartridge, it may not support the standard way of saving data.
Also see this article: https://flashcartdb.com/index.php/Clone_and_Repro_Cart_Problems

Bullet55NL, 2023-01-04:
Hi! And thanks for responding so quickly. I got it somewhat working but when I try to open the save (which is now able to press when I shutdown my GBC) it says 'save file corrupted'

LucentW, 2023-01-04:
> Hi! And thanks for responding so quickly. I got it somewhat working but when I try to open the save (which is now able to press when I shutdown my GBC) it says 'save file corrupted'

As @lesserkuma said, if the cart has no battery there's no way it's going to hold the save except by patching the ROM to reflash it back into the game flash, which is out of scope of the FlashGBX project.

I made a template of one of the patches the bootleggers use, but it might not be relevant to your flash chip: https://gist.github.com/LucentW/0aa8e1530f37441a622212689971193e
On the same snippet you can find the link to BennVenn's video explaining how to apply it.

Bullet55NL, 2023-01-04:
> As @lesserkuma said, if the cart has no battery there's no way it's going to hold the save except by patching the ROM to reflash it back into the game flash, which is out of scope of the FlashGBX [...]

Thanks Lucent. I will take a look at this. Seems promising but feels like a lot of work.. Not sure if I am capable enough..

lesserkuma, 2023-01-10:
Closed as the issue is not directly related to FlashGBX.

closed 2023-01-10 by lesserkuma