Lesserkuma/FlashGBX #66: Auto-detect for Ferrante Crafts 512KB

Anonymous941, 2023-01-09:
I got a Ferrante Crafts 512KB and FlashGBX cannot auto-detect it, but it works when I select it manually.  Perhaps the app can see if the audio pin is writable, and if so check the size and automatically choose this cartridge type.

lesserkuma, 2023-01-12:
If I ever get my hands on one, I can take a closer look.

Anonymous941, 2023-01-12:
Let me know if I can help, I've got one and can do any testing you need.

lesserkuma, 2023-01-12:
Can you post the detailed results that appear when you click Detect Flash Cart?

Anonymous941, 2023-01-12:
* Real time clock: Not available
* ROM size: 128 KB
* Save type: 256 SRAM (32 KB) (this is incorrect, it can't save or even be flashed by the gameboy)
* Mapper type: MBC5+SRAM+BATTERY (it doesn't have a battery)
* Cart: Generic ROM Cartridge

lesserkuma, 2023-01-12:
I actually mean something like this:
```
Game Title: TEST
ROM Size: 8 MB
Save Type: 1M SRAM (128 KB)
Mapper Type: Default (MBC5)

Flash ID Check:
[     ROM     ] C3 AF 03 00 00 00 00 00 
[AUDIO/ 555/A9] 89 89 15 15 00 00 00 00 
[AUDIO/5555/A9] 89 89 15 15 00 00 00 00 
[AUDIO/ AAA/A9] 89 89 15 15 00 00 00 00 
[AUDIO/4555/A9] 89 89 15 15 00 00 00 00 
[AUDIO/7555/A9] 89 89 15 15 00 00 00 00 
[AUDIO/4AAA/A9] 89 89 15 15 00 00 00 00 
[AUDIO/7AAA/A9] 89 89 15 15 00 00 00 00 
[AUDIO/   0/90] 89 89 15 15 00 00 00 00 

Common Flash Interface Data:
Device size: 0x0800000 (8.00 MB)
Voltage: 4.5–5.5 V
Single write: True
Buffered write: True (32 Bytes)
Sector erase: 1024–16384 ms
Sector flags: 0x89
Region 1: 0x0000000–0x07FFFFF @ 0x20000 Bytes × 64

Compatible Cartridge Types:
- BUNG Doctor GB Card 64M*
- Mr Flash 64M

FlashGBX v3.20 | GBxCart RW v1.4a – Firmware R38+L8
```
This should appear if you click [Detect Flash Cart]!

Anonymous941, 2023-01-12:
```plain
The following cartridge configuration was detected:

Game Title: ...
Cartridge Type: Generic ROM Cartridge (not rewritable or not auto-detectable)
```

lesserkuma, 2023-01-12:
Hmm, that’s odd. FlashGBX does fully support the audio port! The example above is also a cartridge that uses audio. So I don’t know... Can you check if there’s continuity from the audio pin to the flash chip?

Anonymous941, 2023-01-12:
It's able to flash successfully when I select the type manually so I don't think that's the issue.  Maybe it's not supported in the auto-detection or there's something different about this type?

lesserkuma, 2023-01-12:
Is there any error or other message displayed in the terminal window during auto-detect or manually selecting and flashing a ROM?

Anonymous941, 2023-01-12:
I'm using the latest version, v1.4a with firmware R38+L8, on an insideGadgets GBxCart RW v1.4a Pro.  [debug.log](https://github.com/lesserkuma/FlashGBX/files/10406773/debug.log)

lesserkuma, 2023-01-12:
Thanks! It looks like your debug.log wasn’t newly generated though. Can you repeat the above steps again? Make sure you definitely hold the shift key on your keyboard while clicking the "Show configuration directory" menu item. It should also say `debug.log written.` in the terminal window if done correctly.

Anonymous941, 2023-01-13:
Sorry about that, [here's the new debug.log](https://github.com/lesserkuma/FlashGBX/files/10412627/debug.log)!

lesserkuma, 2023-01-13:
Thank you! Ok, click Config → Show configuration directory to open the config folder again, then put the following file into that folder (overwrite the existing file), then restart FlashGBX.
[fc_DMG_SST39SF040_AUDIO.txt](https://github.com/lesserkuma/FlashGBX/files/10414564/fc_DMG_SST39SF040_AUDIO.txt)

Does that fix it?

Anonymous941, 2023-01-14:
It's now being detected as a DIY cart:

> The following cartridge configuration was detected:
>
> ROM Size: 512 KB
> Mapper Type: Manual selection
> Cartridge Type: DIY cart with SST39SF040 @ AUDIO (or compatible)

lesserkuma, 2023-01-14:
Sounds good. It should also say `Ferrante Crafts cart 512 KB` if you click the Details button.

Anonymous941, 2023-01-15:
In the dropdown, Ferrante Crafts cart 512 KB is a separate option then what it selects, is it still safe to flash using this setting?

lesserkuma, 2023-01-15:
Yep! These two just function as aliases between another. They’re the same :)

Anonymous941, 2023-01-16:
Okay, thanks for fixing it!  Hopefully this doesn't happen to anyone else.

closed 2023-01-16 by Anonymous941