Lesserkuma/FlashGBX #67: Can I use this cart?

ccs21, 2023-01-27:
There are no words other than DRV written on the board.

Running the detect gave me the following results:

Attach photos together.

`The following cartridge configuration was detected:

Game Title: POKEMON_SLV
Save Type: 1M SRAM (128 KB)

Flash ID Check:
[     ROM     ] F3 C3 00 01 00 00 00 00 
[WR   /4AAA/A9] 02 02 4A 4A 00 00 00 00 
[WR   /7AAA/A9] 02 02 4A 4A 00 00 00 00 

Common Flash Interface Data:
Swapped pins: (0, 1)
Device size: 0x0200000 (2.00 MB)
Voltage: 2.7–3.6 V
Single write: True
Buffered write: False
Sector erase: 1024–16384 ms
Region 1: 0x0000000–0x0003FFF @ 0x4000 Bytes × 1
Region 2: 0x0004000–0x0007FFF @ 0x2000 Bytes × 2
Region 3: 0x0008000–0x000FFFF @ 0x8000 Bytes × 1
Region 4: 0x0010000–0x01FFFFF @ 0x10000 Bytes × 31

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board.

FlashGBX v3.21 | GBxCart RW v1.4a – Firmware R39+L8
`

[image]

[image]

[image]

[image]

[image]

[image]

lesserkuma, 2023-01-27:
Thanks! Can you try if this option would work?
`SD007_48BALL_64M with GL032M11BAIR4`
Manually select it from the list, then click Write ROM. The maximum ROM size is 2 MB.

ccs21, 2023-01-31:
It Works!

Thanks a lot!!!!

lesserkuma, 2023-01-31:
Nice! I’ll add it to the auto-detect for the next release! ^^

closed 2023-01-31 by lesserkuma