Lesserkuma/FlashGBX #69: FlashGBX No device found when trying to connect.

devicemodder, 2023-02-16:
Hello, I have a GBxCart RW V1.4 that i am trying to use with the latest flashGBX software. The issue is that the software can't see the GBx board. it shows up in device manager as COM1, so i dunno where the issue is. Running windows 11 and installed the drivers for the CH340.

This worked previously on my windows 10 machine with FlashGBX 2.8.

I am running the latest version of FlashGBX as of this post.

lesserkuma, 2023-02-16:
Hello! Can you post a photo of your GBxCart RW device so the exact model can be determined?
You can also join insideGadgets’ Discord at https://gbxcart.com/discord for hardware and driver troubleshooting.

devicemodder, 2023-02-16:
I'll post when I get home from work

devicemodder, 2023-02-16:
[image]
[image]

lesserkuma, 2023-02-16:
Seems fine and I see no damage. Can you try some different USB data cables and USB ports? Just to be sure!

devicemodder, 2023-02-16:
I only have one other cable with me at the moment, tried it and i get the same... 

hrm... device manager has a bunch of bluetooth serial devices. i disable them and get: 

[image]

devicemodder, 2023-02-16:
COM3 is the GBx Cart RW

lesserkuma, 2023-02-16:
I think after pulling a list of all devices from the operating system, it detected the GBxCart RW on COM3, but when trying to actually connect to it, it’s no longer there. Might be a loose connection or otherwise unstable. I’d try more different cables and different USB ports. Also give this software a try: https://shop.insidegadgets.com/wp-content/uploads/2018/05/GBxCart_RW_GUI_Interface_v1.49.zip

If it still doesn’t work even with the other GUI software, it should be unrelated to this software and more likely an issue with the device, cables, drivers, operating system or USB controller. Please contact insideGadgets in that case.

devicemodder, 2023-02-16:
Ok, I'll check it out.

closed 2023-03-04 by lesserkuma

happysoul, 2023-10-24:
Maybe your GBxCart has a broken chip （6pin use for switch 5v 3.3v）
When you connect cart to GBxCart，Cart pin 1 may connect to GBxCart pin 1 & 2
pin1 = vcc , 3.3v
pin2 = clk , connect to GND on GBxCart

when connect/disconnect cart to GBxCart 
it happens looks like this |/|  
board pin | | 
cart pin / 
If short circuit ，GBxCart chip burn down. atmega8515 may not work because of vcc pin got 100Ω to GND

Test GBxCart pin1, if it is 3.3v ,not gnd