Lesserkuma/FlashGBX #71: Add profile for Fujitsu branded 29F106

sillyhatday, 2023-03-25:
I have come across some Fujitsu branded chips for nearly half the price of AMD versions. After some tinkering you can get them to flash properly using the generic settings suggested. Could there be support to have its own profile properly auto detected?

[image]

This is what is brought up after using Detect Flash Cart function below.

[image]

lesserkuma, 2023-03-25:
OK, if you could show me a picture of the chip with visible markings, I will add it in the next release!

sillyhatday, 2023-03-26:
[image]

I hope this is clear enough. If not I can wait until there is daylight to send another

sillyhatday, 2023-03-26:
[image]

sillyhatday, 2023-03-26:
[image]

closed 2023-04-06 by lesserkuma