Lesserkuma/FlashGBX #72: Support Request for Reflashing Action Replay Max Duo

WilliamDingler, 2023-04-04:
I have a corrupted Action Replay Max Duo. I am wondering if reflashing the SST39VF800A would be possible with FLASHGBX and a GBxCart RW. Would It be possible to add support for fully reflashing bricked GBA action replays?

I have a GBxCart rw v1.4a on Firmware R39+l8 and I am using FlashGBX v3.22. Auto-detecting the device hasn't been successful. Full backups of the SST39VF800A exist here.  https://github.com/RWeick/REF1714-NDS-GBA-Action-Replay-Max/commits?author=RWeick

WilliamDingler, 2023-04-04:
[image]

WilliamDingler, 2023-04-04:
[image]

WilliamDingler, 2023-04-04:
[image]

LucentW, 2023-04-04:
Isn't it already supported? Pick the `Action Replay Ultimate Codes (SST39VF800A)` option from the dropdown and you're good to go!

WilliamDingler, 2023-04-04:
> Isn't it already supported? Pick the `Action Replay Ultimate Codes (SST39VF800A)` option from the dropdown and you're good to go!

I have tried a few different times to no avail. I have tried both bin files in the reference repo and the official 1.6 update file. I even thought the switch position would make a difference. Here is a screen shot of the error.
[image]

lesserkuma, 2023-04-05:
If I ever come across a working (non-corrupted) one, I will do my best and add support!