Lesserkuma/FlashGBX #74: Unable to detect flash cart types

inmortalkaktus, 2023-04-27:
Hello. 

I've just received my GBxCart v1.4a, thus, I've downloaded FlashGBX and updated the firmware.

I've been trying to flash some of the 6 GBA cartridged I do have, but I had no luck, none of them are recognized by the program. I've tried like 20 times to unplug and plug again with no luck. Both, my GBA and my DS Fat gets all the cartirdges as much, at the third intent.

I've tried to take a look on their PCB to get the names manually, but they're not listed.

SHREK THE THIRD (which I've seen in the db_AGB.json file)
[image]

SONIC ADVANCE + CHUCHU ROCKET
[image]

POKEMON SHAPPIRE (ESP)
[image]

There are some of the cartridges I've wanted to play, and I got surprised that even Pokemon wasn't recognised properly. ¿What could be? I've tried to write selecting some random chipsets on a game that I didn't dare to lose, but I always get an error.

Is there some way to know if there is some item on the list that could match good for the chipsets I own?

Thanks in advance!

LucentW, 2023-04-27:
Remember that original cartridges don't hold the game on a flash chip but actually mask ROM which is "etched" during the manufacturing phase and can't be altered. You need either actual flashcarts (i.e. insideGadgets') or cheap bootlegs off the usual suspects, namely AliExpress, which happen to be 99% flashable from the carts' edge connector.

If you're looking to dump and preserve your games and saves, make sure to clean the pins on your carts with either a pencil eraser or a qtip dabbed on some rubbing alcohol.

lesserkuma, 2023-04-27:
@LucentW explained it very nicely! These are genuine game cartridges by Nintendo so you will be limited to **Backup ROM**, **Backup Save Data** and **Write Save Data** functions. If these functions don’t work correctly, clean the pins thoroughly as explained above. If you still have trouble, please attach screenshots of any errors you encounter.

closed 2023-05-02 by lesserkuma