Lesserkuma/FlashGBX #76: Error when starting GUI on OSX

LouWii, 2023-05-03:
Hi there,

I've installed the package in pip, everything went fine.

```
Installing collected packages: shiboken2, pyserial, urllib3, python-dateutil, PySide2, idna, charset-normalizer, certifi, requests, FlashGBX
Successfully installed FlashGBX-3.26 PySide2-5.13.2 certifi-2022.12.7 charset-normalizer-3.1.0 idna-3.4 pyserial-3.5 python-dateutil-2.8.2 requests-2.29.0 shiboken2-5.13.2 urllib3-1.26.15
```

When I try to run the command to start FlashGBX, an error shows up about failing to start the GUI. I'm unsure why since pip installed everything properly.

```
NOTE: GUI mode couldn’t be launched, but the application can be run in CLI mode.
      Optional command line switches are explained above.

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/FlashGBX/FlashGBX.py", line 187, in main
    app = FlashGBX_GUI.FlashGBX_GUI(args)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/FlashGBX/FlashGBX_GUI.py", line 43, in __init__
    self.setWindowFlags(self.windowFlags() | QtCore.Qt.MSWindowsFixedSizeDialogHint)
TypeError: 'PySide2.QtCore.Qt.WindowType' object cannot be interpreted as an integer
```

* OSX 13.3.1
* pip 22.3.1 from /usr/local/lib/python3.11/site-packages/pip (python 3.11)
* Python 3.11.2

lesserkuma, 2023-05-03:
Hi, can you try again with PySide6?
`pip3 uninstall PySide2`
`pip3 install PySide6`
And then run FlashGBX again.

LouWii, 2023-05-04:
Yup, that worked! Thanks!

Does that mean I should have used the second install command `pip3 install "FlashGBX[qt6]"`? Instead of `pip3 install "FlashGBX[qt5]"`

lesserkuma, 2023-05-04:
Yes, but there was also a wrong value in my setup.py that caused the qt5 one to download an old version of PySide2. They don't let you edit a release once it's been published, so this issue will also be fixed in the next release.

LouWii, 2023-05-05:
Alright. Thanks again for the help.

closed 2023-05-05 by LouWii