Lesserkuma/FlashGBX #79: Support request for Flash Advance Card Xtreme 512

8Gilles8, 2023-05-28:
Hello, I'm trying to flash my Flash Advance Card Xtreme 512.

I've used a 32 MiB to test flashing.

The "Generic Flash Cartridge (0/90)" config tells me that ROM
writing is complete after only 0.50 seconds and the card is left 
apparently unchanged.

I tryed to flash it like a "Flash2Advance 256M with 2x 28F128J3A150".
FlashGBX seems to erase the first sector, but after about 12 seconds,
I get a message saying an error occured while writing 0x40000 bytes
at position 0x0 (0 Bytes).
Once again, the card is left unchanged.

Please, can you find a config to make this flash card compatible ?

Thanks in advance.

Here are results of card analysis and photos :

The following cartridge configuration was detected:

ROM Title: POGOSHELL
Save Type: Unlicensed 512K SRAM (64 KiB)

Flash ID Check:
[   ROM   ] 2E 00 89 00 24 FF 02 88 
[   555/AA] 89 00 00 EA 02 88 AE 51 
[   555/A9] 89 00 00 EA 02 88 AE 51 
[  5555/AA] 89 00 00 EA 02 88 AE 51 
[  5555/A9] 89 00 00 EA 02 88 AE 51 
[  4555/AA] 89 00 00 EA 02 88 AE 51 
[  4555/A9] 89 00 00 EA 02 88 AE 51 
[  7555/AA] 89 00 00 EA 02 88 AE 51 
[  7555/A9] 89 00 00 EA 02 88 AE 51 
[     0/90] 89 00 89 00 02 88 02 88 

Common Flash Interface Data: No data provided

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (0/90)” a try at your own risk.

FlashGBX v3.29 | GBxCart RW v1.4a – Firmware R42+L10

[image]

[image]

lesserkuma, 2023-05-28:
Hi! It might need special unlock sequences. If you have the original flasher and a 32 channel logic analyzer, please capture the signals while writing a small ROM and send me the captured file.

8Gilles8, 2023-05-29:
Hi,

Thanks for your answer.

I still have the original linker. It's a [flash linker xtreme](https://www.gameboy-advance.net/flash_linker/gba_flash_linker_xtreme_usb.htm). But I don't have access to an analyzer.

Here is the software I use :
[XtremeLink (Soft & Drivers).zip](https://github.com/lesserkuma/FlashGBX/files/11589150/XtremeLink.Soft.Drivers.zip)
It needs to run under windows xp (that's why I wanted a more modern linker).
The sources seems to be closed, so it shouldn't help a lot.

The same linker and software can be used to flash a flashcard advance 256M: [it's the first photo on this page](https://www.gameboy-advance.net/flash_card/flash_advance_256m_card.htm).
This latest card seems to flash correctly with FlashGBX v3.29 and GBxCart RW v1.4a – Firmware R42+L10.

If you know the linker, there's a 10pin header on the board. I have a USB<->Serial device I use with an Arduino Mini pro.
Maybe there's a way to get what you need with this, but I need to know how to wire it correctly and how to access serial communication. I can use Serial monitor from Arduino IDE, but I'd need to know speed, parity, etc...
I can also run a python script if needed.

I can even solder if the onboard header is not suitable. But I'd need very precise directions. If there's no way to use FlashGBX with this card, I don't want to fry the original linker.

That's all I can think of right now.