Lesserkuma/FlashGBX #80: Unknown flash cartridge type

odecif, 2023-05-31:
Hello,

I've been trying to (successfully) dump Golden Sun AGB-AGSE-USA but every time I make a dump the checksum doesn't match the one in the database. So now I tried the "Analyze Flash Cart" -button which came back like this:

```
The following cartridge configuration was detected:

ROM Title: Golden_Sun_A
Save Type: Unlicensed 512K SRAM (64 KiB)

Flash ID Check:
[   ROM   ] EE 00 00 EA 24 FF AE 51 
[   555/A9] 6B B8 00 EA 24 FF AE 51 
[  5555/A9] 8D 46 00 EA 24 FF AE 51 
[   AAA/A9] 8A E0 00 EA 24 FF AE 51 
[  AAAA/A9] C8 5C 00 EA 24 FF AE 51 
[  4555/AA] 0C 4B 00 EA 24 FF AE 51 
[  4555/A9] 0C 4B 00 EA 24 FF AE 51 
[  7555/AA] 00 04 00 EA 24 FF AE 51 
[  7555/A9] 00 04 00 EA 24 FF AE 51 
[  4AAA/AA] 0C 4B 00 EA 24 FF AE 51 
[  4AAA/A9] 0C 4B 00 EA 24 FF AE 51 
[  7AAA/AA] 00 04 00 EA 24 FF AE 51 
[  7AAA/A9] 00 04 00 EA 24 FF AE 51 
[     0/90] 00 04 00 EA 24 FF AE 51 

Common Flash Interface Data: No data provided

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (0/90)” a try at your own risk.

FlashGBX v3.29 | GBxCart RW v1.4a – Firmware R42+L10
```

The dump plays (in VBA-M) with some minor artifacts mid- and late-game (one Djinni "Quartz" looks distorted and the floor mat in the ship that crosses the Karagol Sea to Kalay hovers above everything and everyone).

I get the same md5sum when trying with either "Generic ROM Cartridge" and " Generic Flash Cartridge(0/90)". I also recently put a new battery in since old one was dead and tried to dump both with and without the battery, but nothing gave me a correct checksum.

I've also used a generous amount of isopropyl alcohol and q-tips.

If there is anything else I can contribute with, please let me know :)

[image]

[image]

The board only reads "G108" and the chips on the board seem to read:

M6MGT321S4TP (mitsubishi-logo) BGO 002B200
M6MGT323S4TP (mitsubishi-logo) BGO 952B201
S2ATP 51208

Regards

LucentW, 2023-05-31:
That looks like an old style bootleg cartridge, which are hit or miss about reflashing them. The option for the flash cart is used only when *flashing*, not when dumping.
Consider that the game has been definitely SRAM patched (the original cart should have 512K flash, not SRAM), hence why the checksum is not matching with the one into the database.

odecif, 2023-05-31:
Wow, so I am actually the owner of an old bootleg cartridge, would you look at that. I actually got the game as a bundle with 2 other games and a GameBoy Advance "ages" ago at a local video game shop (talking year ~2002) so I'm actually surprised its bootleg.

Ah, so the Cart-option is only for _flashing_, got it.

Thanks for the quick response!

closed 2023-05-31 by odecif