Lesserkuma/FlashGBX #81: mbc5 cart with 64MB,will cause a bug.

simonkwng, 2023-06-10:
I make a DIY mbc5 cart with MX29lv640 and found a buger when I flash some rom to it. 
the rom if the data in 0x3FFF address is not zero, the buger will breakout. as the picture show.  
[image]

and I found the reason:
when write the frist 16K dates to the flash chip, the address is 0-0x3FFF. and the mbc5 high bit rombank  reg address is 0x3000-0x3FFF. so any data is not 0 in 0x3000-0x3FFF will cause the highbit  rombank  addr switch to 1. the data in last addr 0x3fff will keep the highbit rombank value until next set rombank cmd which has two cmds[0x2000(low 8bit) and 0x3000 (high bit )].

flashgbx set the rombank addr use two 2 cmds only in index ==0 or >256 in mbc5 mapper.  actually  the index need <=1 or > 256. or all rombank use two cmds.
[image]
[image]

sorry ,my poor english

simonkwng, 2023-06-12:
I try to modified the code as follow, then flashgbx works fine! : )
[image]

lesserkuma, 2023-06-27:
Please check with version 3.31 if the issue is fixed, thanks! ^^