Lesserkuma/FlashGBX #82: Can this reproduction cartridges be support?

kane159, 2023-06-17:
The following cartridge configuration was detected:

ROM Title: BOKUJOU GB2

Flash ID Check:
[     ROM     ] 87 F5 3E 00 8C 67 F1 85 
[WR   /7AAA/AA] 00 00 00 01 00 02 00 01 
[WR   /7AAA/A9] 00 00 00 01 00 02 00 01 
[WR   /   0/90] 00 00 00 01 00 02 00 01 
[AUDIO/ 555/A9] 00 00 00 01 00 02 00 01 
[AUDIO/ 555/AA] 00 00 00 01 00 02 00 01 
[AUDIO/5555/A9] 00 00 00 01 00 02 00 01 
[AUDIO/5555/AA] 00 00 00 01 00 02 00 01 
[AUDIO/ AAA/A9] 00 00 00 01 00 02 00 01 
[AUDIO/ AAA/AA] 00 00 00 01 00 02 00 01 
[AUDIO/4555/A9] 00 00 00 01 00 02 00 01 
[AUDIO/4555/AA] 00 00 00 01 00 02 00 01 
[AUDIO/7555/A9] 00 00 00 01 00 02 00 01 
[AUDIO/7555/AA] 00 00 00 01 00 02 00 01 
[AUDIO/4AAA/A9] 00 00 00 01 00 02 00 01 
[AUDIO/4AAA/AA] 00 00 00 01 00 02 00 01 
[AUDIO/7AAA/A9] 00 00 00 01 00 02 00 01 
[AUDIO/7AAA/AA] 00 00 00 01 00 02 00 01 
[AUDIO/   0/90] 00 00 00 01 00 02 00 01 
[WR   / 555/AA] 00 00 00 01 00 02 00 01 
[WR   /5555/AA] 00 00 00 01 00 02 00 01 
[WR   / AAA/AA] 00 00 00 01 00 02 00 01 
[WR   /4555/AA] 00 00 00 01 00 02 00 01 
[WR   /7555/AA] 00 00 00 01 00 02 00 01 
[WR   /4AAA/AA] 00 00 00 01 00 02 00 01 

Common Flash Interface Data: No data provided

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (WR/AAA/AA)” a try at your own risk.

FlashGBX v3.30 | GBxCart RW v1.3 – Firmware L1

and here is the cartridge’s circuit board
https://imgur.com/a/yw5WqVL

lesserkuma, 2023-06-17:
Hi! The flash memory doesn't appear to respond in a meaningful way. If it's still the same numbers even after cleaning the contacts, it's likely not writable through the cartridge edge.

kane159, 2023-06-17:
> Hi! The flash memory doesn't appear to respond in a meaningful way. If it's still the same numbers even after cleaning the contacts, it's likely not writable through the cartridge edge.

ok thats unfortunately...i have two this type of cart.......then how about this one? this one is a very old cart i found in my old home

The following cartridge configuration was detected:

ROM Title: YUGIOUDM4J
Save Type: 512K SRAM (64 KiB)

Flash ID Check:
[     ROM     ] D9 00 00 00 00 00 00 00 
[WR   /7555/AA] C8 3E 00 EA DC C8 F1 C9 
[WR   /7555/A9] C8 3E 00 EA DC C8 F1 C9 
[WR   /7AAA/AA] D8 38 00 E8 D8 C8 F0 C8 
[WR   /7AAA/A9] D8 38 00 E8 D8 C8 F0 C8 
[AUDIO/ 555/A9] D8 00 00 00 00 00 00 00 
[AUDIO/ 555/AA] D8 00 00 00 00 00 00 00 
[AUDIO/5555/A9] D8 00 00 00 00 00 00 00 
[AUDIO/5555/AA] D8 00 00 00 00 00 00 00 
[AUDIO/ AAA/A9] D8 00 00 00 00 00 00 00 
[AUDIO/ AAA/AA] D8 00 00 00 00 00 00 00 
[AUDIO/4555/A9] D8 00 00 00 00 00 00 00 
[AUDIO/4555/AA] D8 00 00 00 00 00 00 00 
[AUDIO/7555/A9] C8 38 00 E8 D8 C8 F0 C8 
[AUDIO/7555/AA] C8 38 00 E8 D8 C8 F0 C8 
[AUDIO/4AAA/A9] D8 00 00 00 00 00 00 00 
[AUDIO/4AAA/AA] D8 00 00 00 00 00 00 00 
[AUDIO/7AAA/A9] D8 38 00 E8 D8 C8 F0 C8 
[AUDIO/7AAA/AA] D8 38 00 E8 D8 C8 F0 C8 
[WR   / 555/AA] D8 00 00 00 00 00 00 00 
[WR   /5555/AA] D8 00 00 00 00 00 00 00 
[WR   / AAA/AA] D8 00 00 00 00 00 00 00 
[WR   /4555/AA] D8 00 00 00 00 00 00 00 
[WR   /4AAA/AA] D8 00 00 00 00 00 00 00 

Common Flash Interface Data: No data provided

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (WR/AAA/AA)” a try at your own risk.

FlashGBX v3.30 | GBxCart RW v1.3 – Firmware L1

lesserkuma, 2023-06-27:
Hmm, that one does not appear to be re-writable through the cartridge edge either ☹️

closed 2023-07-25 by lesserkuma