Lesserkuma/FlashGBX #83: New cartridge type: 32mbit 256kbit SRAM MBC5 from gamebank-web

zipplet, 2023-07-10:
Cartridge was obtained from a Japanese company a long time ago.
Flash: 32mbit
SRAM: 256kbit
Mapper: MBC5

Here's the info:
```
The following cartridge configuration was detected:

ROM Title: CGBWARIOLAND2
Save Type: 256K SRAM (32 KiB)

Flash ID Check:
[     ROM     ] 0C 00 92 00 BC DA 9F 00 
[WR   / AAA/AA] 20 20 56 56 00 00 01 01 
[WR   /4AAA/AA] 20 20 56 56 00 00 01 01 
[WR   /7AAA/AA] 20 20 56 56 00 00 01 01 

Common Flash Interface Data:

Device size: 0x0400000 (4.00 MiB)
Voltage: 2.7–3.6 V
Single write: True
Buffered write: False
Sector erase: 1024–8192 ms
Sector flags: Reversed (0x03)
Region 1: 0x0000000–0x000FFFF @ 0x2000 Bytes × 8
Region 2: 0x0010000–0x03FFFFF @ 0x10000 Bytes × 63

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (WR/AAA/AA)” a try at your own risk.

FlashGBX v3.31 | GBxCart RW v1.3 – Firmware L1
```

[image]

zipplet, 2023-07-10:
Using "Generic Flash Cartridge (WR/AAA/AA)" did seem to write it (I clicked "3.3V" at the popup for the voltage choice).

lesserkuma, 2023-07-10:
Thanks! I will add it to the auto-detect routine in the next release!

closed 2023-07-17 by lesserkuma