Lesserkuma/FlashGBX #89: When i launch FlashGBX，no gui and display this error

SunnyIMG, 2023-10-27:
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1131)'))': /simple/pyside2/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1131)'))': /simple/pyside2/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1131)'))': /simple/pyside2/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1131)'))': /simple/pyside2/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1131)'))': /simple/pyside2/
Could not fetch URL https://pypi.org/simple/pyside2/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pyside2/ (Caused by SSLError(SSLEOFError(8, 'EOF occurred in violation of protocol (_ssl.c:1131)'))) - skipping
ERROR: Could not find a version that satisfies the requirement PySide2==5.15.2.1 (from versions: none)
ERROR: No matching distribution found for PySide2==5.15.2.1

happysoul, 2023-11-10:
缺少依赖包，你网络连不上下载源

Your network cann't connect  https://pypi.org/simple/pyside2/

# switch download source 切换源（清华、阿里）
pip config set global.index-url https://mirrors.aliyun.com/pypi/simple/
pip config set global.index-url https://pypi.tuna.tsinghua.edu.cn/simple/

lesserkuma, 2023-11-16:
Looks like your internet service provider is blocking the official Python Package Index (pypi.org). I'm afraid I have no solution for this, but you may try what happysoul suggested.

closed 2023-11-16 by lesserkuma