Lesserkuma/FlashGBX #91: RTC ROM Read Issue When Backing Up Save on Authentic Pokemon Emerald

Xeavior, 2023-11-03:
When using Backup Save data on my Pokemon Emerald, the save properly backs up. However, for some time after this, the cartridge refuses to read properly. It continues to claim the pins are dirty or not aligned. More interestingly, it seems that once the cart can be read again, the RTC has reset to 2000-01-01  00:00:00 and it works and reads as normal. During the time it takes to return to normal, the flash cart analyzer continuously returns different information shown after the photos of the cartridge.

The steps I take for this to happen:

- I plug in my GbxCart RW v1.4b Pro to my PC
- I Open FlashGBX v3.34
- I Insert the game cartridge into the GbxCart RW v1.4b Pro
- I select "Connect" in FlashGBX
- I select "Game Boy Advance" Radio Button in FlashGBX
- I verify it read the cartridge properly
- I select "Backup Save Data" Button in FlashGBX
- I save the .sav file
- I select "Disconnect" in FlashGBX
- I remove the cartridge from my GbxCart RW v1.4b Pro

[image]
[image]

```
The following cartridge configuration was detected:

ROM Title: _popo
Save Type: Unlicensed 1M SRAM (128 KiB)

Flash ID Check:
[   ROM   ] 00 EA 48 4A 24 FF AC 51 
[   555/AA] FD 52 48 4A 24 FF AC 51 
[  5555/AA] FD 52 48 4A 24 FF AC 51 
[   AAA/AA] FD 52 48 4A 24 FF AC 51 
[   AAA/A9] FD 52 48 4A 24 FF AC 51 
[     0/90] FD 52 48 4A 24 FF AC 51 

Common Flash Interface Data: No data provided

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (0/90)” a try at your own risk.

FlashGBX v3.34 | GBxCart RW v1.4a/b – Firmware R42+L10
```

```
The following cartridge configuration was detected:

ROM Title: _popo
Save Type: Unlicensed 1M SRAM (128 KiB)

Flash ID Check:
[   ROM   ] 00 EA 48 4A 24 FF AC 51 
[  5555/A9] 00 EA 24 FF 24 FF AC 51 
[   AAA/A9] 7D 00 00 EA 24 FF AC 51 
[  4555/A9] 00 EA 24 FF 24 FF AC 51 
[  7555/A9] 7D 00 00 EA 24 FF AC 51 
[  4AAA/A9] 7D 00 00 EA 24 FF AC 51 
[  7AAA/A9] 00 EA 24 FF 24 FF AC 51 

Common Flash Interface Data: No data provided

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (AAA/A9)” a try at your own risk.

FlashGBX v3.34 | GBxCart RW v1.4a/b – Firmware R42+L10
```
[image]

lesserkuma, 2023-11-03:
Hi! Just tested my Pokémon Emerald and couldn’t reproduce this. It could be a hardware problem with the GBxCart RW device or the cartridge, but also a connectivity/voltage problem etc. Can you try a few different USB cables and also a different computer and see if the issue persists?

Xeavior, 2023-11-03:
Thanks for the quick response! I am unsure of diagnosing the GBxCart. However, since my NDS does not recognize the cart after the transfer I would assume something is happening to the cart after a save is pulled. Over time, eventually both my NDS and the GBxCart are able to read the cartridge once again.

With it in its current bad state, a different port and different cable return back the same results.
I did not notice any immediate trace issues on the Pokemon Emerald or the GBxCart.

I also could not find anything online on how an RTC could affect reading ROM like I am experiencing. The fact that over time it works again, however, makes me feel like that is where the issue may lie.

Does your Pokemon Emerald also have a replaced battery? Or, is the battery dead/removed?

[image]

Xeavior, 2023-11-03:
I have decided this indeed may be a hardware issue with the game cart. Closing the ticket.

closed 2023-11-03 by Xeavior