Lesserkuma/FlashGBX #92: Issue with save files being ripped as corrupt files

Marka1193, 2023-11-15:
I've updated to the new version, and did the update for the new firmware for the insideGadgets GBxCart RW v1.4a Pro.

Now everytime I try and take a backup copy of the Pokémon games I've got (Red, Blue, Yellow, Gold, Silver, Crystal), the program says it was successful. But whenever I try and load them into an Emulator and PKHeX, the save files keep being flagged as corrupt files, and neither can read them. I used VBA, and several retroarch emulators to test them.

The reader and program are creating corrupt files, that they say are readable files.

Before I did the updates, I was able to take copies of the files successfully that weren't corrupt.

lesserkuma, 2023-11-16:
Hi, I just double checked with the latest version of FlashGBX and my copies of Pokémon Red, Blue, Yellow, Gold, Silver and Crystal and I can't reproduce this issue. It could be related to your cartridges (connectivity problems or their batteries being dead), or a problem with the GBxCart RW hardware. Clean the pins of the cartridge in question and see if you can backup the ROM successfully (no checksum errors) and _then_ backup the save data. If it still doesn't work, you can join the [insideGadgets discord](https://gbxcart.com/discord) or contact their support directly. It's also possible to downgrade the software and firmware if need be.

Marka1193, 2024-01-20:
Hi again, I've recently updated to the latest version of FlashGBX, and it's refusing to read any cart. all of the carts have been cleaned, and all the internal batteries are fresh. The firmware of the reader is up to date too. Apparently according to the program and reader, the games aren't in the game database, and my copy of Silver is giving the same error, saying in need to clean the pins. This is what it's saying;
[image]

It's making save files again that can't be read by anything.

lesserkuma, 2024-01-20:
The error message you’re seeing appears if there’s a mismatch between two identical reads from ROM, one after another. This usually happens only if the cartridge contacts and/or GBxCart RW slot need thorough cleaning with 99.9% IPA.
As this is a hardware issue, this is not something I can help with. You can contact AlexiG on the [insideGadgets discord](https://gbxcart.com/discord) for proper customer support for the device.