Lesserkuma/FlashGBX #94: [REQUEST] High DPI GUI

Clads92, 2023-11-22:
Hi,

Thanks for this awesome work, but on my 4K screen the gui is very tiny, it's possible to get high dpi feature for a future version please ?
[image]

lesserkuma, 2023-11-22:
Hi! Is this a computer with just one display connected or do you have multiple displays? If it's more than one display, do they run on different DPI?

Clads92, 2023-11-22:
It's a mono display (LG cx oled)
Thanks for your reply

lesserkuma, 2023-11-22:
I think it's a bug with the GUI framework ([PySide2](https://pypi.org/project/PySide2/)).
If you're comfortable installing [Python](https://www.python.org/downloads/) on your system, you can then enter this in a Terminal window:
```cmd
pip install "FlashGBX[qt6]"
python -m FlashGBX
```
This would make it use the newer PySide6 framework instead. I hope this works better.

Clads92, 2023-11-23:
Hi, worked like a charm, thanks

lesserkuma, 2023-11-23:
Sounds good! I'll try to add a PySide6/Qt6 portable version for the next release.

closed 2023-11-23 by lesserkuma