Lesserkuma/FlashGBX #95: Unknown cartridge? Beast Wars Japan

2ndmax, 2023-12-09:
Hello! :-)

Please excuse me if this is not the correct channel for my matter.

I brought back a Gameboy Color game from Japan called Beast Wars. The cartridge is not recognized. Because I am quite new to retro gaming, I do not know if this is a mostly unknown cartridge or if it is just broken. In case it is unknown, here are some information:

[image]

[image]

[image]

The following cartridge configuration was detected:

ROM Title: @OKGOONOCLDAAHH
Save Type: 256K SRAM (32 KiB)

Flash ID Check:
[     ROM     ] 73 53 12 13 12 12 12 12 
[WR   / 555/AA] 61 41 00 01 00 00 00 00 
[WR   / 555/A9] 61 41 00 01 00 00 00 00 
[WR   /5555/AA] 61 41 00 01 00 00 00 00 
[WR   /5555/A9] 61 41 00 01 00 00 00 00 
[WR   / AAA/AA] 61 41 00 01 00 00 00 00 
[WR   / AAA/A9] 61 41 00 01 00 00 00 00 
[WR   /4555/AA] 61 41 00 01 00 00 00 00 
[WR   /4555/A9] 61 41 00 01 00 00 00 00 
[WR   /7555/AA] 61 41 00 01 00 00 00 00 
[WR   /7555/A9] 61 41 00 01 00 00 00 00 
[WR   /4AAA/AA] 61 41 00 01 00 00 00 00 
[WR   /4AAA/A9] 61 41 00 01 00 00 00 00 
[WR   /7AAA/AA] 61 41 00 01 00 00 00 00 
[WR   /7AAA/A9] 61 41 00 01 00 00 00 00 
[WR   /   0/90] 61 41 00 01 00 00 00 00 
[AUDIO/ 555/A9] 61 41 00 01 00 00 00 00 
[AUDIO/ 555/AA] 61 41 00 01 00 00 00 00 
[AUDIO/5555/A9] 61 41 00 01 00 00 00 00 
[AUDIO/5555/AA] 61 41 00 01 00 00 00 00 
[AUDIO/ AAA/A9] 61 41 00 01 00 00 00 00 
[AUDIO/ AAA/AA] 61 41 00 01 00 00 00 00 
[AUDIO/4555/A9] 61 41 00 01 00 00 00 00 
[AUDIO/4555/AA] 61 41 00 01 00 00 00 00 
[AUDIO/7555/A9] 61 41 00 01 00 00 00 00 
[AUDIO/7555/AA] 61 41 00 01 00 00 00 00 
[AUDIO/4AAA/A9] 61 41 00 01 00 00 00 00 
[AUDIO/4AAA/AA] 61 41 00 01 00 00 00 00 
[AUDIO/7AAA/A9] 61 41 00 01 00 00 00 00 
[AUDIO/7AAA/AA] 61 41 00 01 00 00 00 00 
[AUDIO/   0/90] 61 41 00 01 00 00 00 00 

Common Flash Interface Data: No data provided

Cartridge Type: Unknown flash cartridge – Please submit the displayed information along with a picture of the cartridge’s circuit board. For ROM writing, you can give the option called “Generic Flash Cartridge (WR/AAA/AA)” a try at your own risk.

FlashGBX v3.35 | GBxCart RW v1.3 – Firmware L1

lesserkuma, 2023-12-09:
Hi! First off, the cartridge is not re-writable as it's a genuine Nintendo game. But the condition of the circuit board seems not so good. Try to clean it thoroughly with some isopropyl alcohol (99%+ if possible), especially the cartridge contacts, and then give it another try. Also confirm whether it starts up on an actual Nintendo Game Boy.

2ndmax, 2023-12-09:
Hey! Thank you for your quick reply. It seems that isopropyl alcohol did do the trick - now it can be read without an issue. Thank you and have a nice day!

closed 2023-12-09 by 2ndmax