Lesserkuma/FlashGBX #98: Support request for installling new version 3.36 on Win7 x64

stevebarn, 2024-01-25:
Hello Lesserkuma, thank you for your great work : )! I tried to update the software to 3.36 on win7 x64 but unfortunately, it cannot be installed correctly, an error message appears "requires API-MS-WIN-CORE-WINRT-L1-1-0.DLL" I tried to install manually the file in both but same error appears again... thank you very much in advance for your help.

P.S. I was running the 3.35 version without any problem before.

lesserkuma, 2024-01-25:
Hi! Sorry for the inconvenience... Can you try the "Qt5" versions from the [release page](https://github.com/lesserkuma/FlashGBX/releases)? This uses slightly older versions of Python, Pillow and the Qt GUI framework and should still be compatible with older Windows versions.

stevebarn, 2024-01-25:
Hello Lesserkuman thank you very much for your great work again and your fast reply : ) I'll give it a try now, could I ask you for what is designed the Qt5 version? Thank you very much again.

lesserkuma, 2024-01-25:
Qt6 is the new version that offers better support for high DPI displays on Windows 11.
Qt5 version is the old version.

I didn't realize switching to Qt6 would break Windows 7 support, so I uploaded a Qt5 build as well.

stevebarn, 2024-01-25:
Thank you for your explanations and your effort to maintain the support for old win versions, I noticed that more and more software and hardware does not still support Win7 and 8, I think about switching to a new version : ) I follow your great work closely, Thank you again for your disponibility and your kindness : )

Last question, do you know and expander for GB/GBC roms like GBATA?

P.S. The Qt5 version works great on win7

closed 2024-01-25 by stevebarn