Lesserkuma/FlashGBX #108: a compatible flash cartiage could not be auto-detected

DrpugSlug, 2024-03-14:
I try to write rom to my gba cartiage but it keep popping this up

LucentW, 2024-03-14:
Can you please send a picture of your GBxCart with the cart plugged in, cartridge (preferably opened to identify the flash chip) and screenshot of the FlashGBX GUI?

DrpugSlug, 2024-03-14:
[image]
[image]

LucentW, 2024-03-14:
You can't reflash original cartridges: they don't have a flash chip but an etched mask ROM.

DrpugSlug, 2024-03-14:
where can i find those

LucentW, 2024-03-14:
All modern cheap bootleg/clone cartridges fit the bill, you can check AliExpress if you want something to fiddle with.
If you're looking for carts i.e. with 1Mbit flash save + real time clock for Gen3 Pokémon and romhacks, you'd better fetch one of insideGadgets' carts that got them both rather than relying on cheap carts.

A good resource to check out with an extensive, but not omnicomprensive (considering bootleggers release new carts every month or so depending on chip stocks), list of flashcarts is [Flash Cart DB](https://flashcartdb.com/index.php/Main_Page).

closed 2024-03-15 by LucentW